fra589/grbl-Mega-5X #301: Cant set steps on y axis

enddDD, 2022-10-15:
Hi i reach to the point in when i need to set the right stepping for my router($100 $101 $102) i managed to set x and z but i am having problems with y. I have double y motors the default steps are 400, if i change that the machine it acts like the steps are changed for only one motor(i think in fact that is happening). Any suggestion how to change to get both motors working correctly? 
Thank you!

RaphaelDives, 2022-10-15:
Hey @enddDD,
You must set the steps for each axis in the order they are configured.
For example:
100 = axis 1 = x
101 = axis 2 = y
102 = axis 3 = z
103 = axis 4 = cloned y
So the values for 101 and 103 has to be the same!
This is the same for all axis parameters. Steps/acceleration/speed/max travel and so on… 😊
Cheers
Raphael

enddDD, 2022-10-15:
Oh, didnt know there is a 103. Thank you!

fra589, 2022-10-15:
Hi @enddDD,  
  
It may be good to read the Wiki: https://github.com/fra589/grbl-Mega-5X/wiki  
I try to keep the documentation up to date, and it takes a lot of time. But it usually answers a lot of questions before they are asked!  
  
Thanks @RaphaelDives for your help :smiley:  
  
@++;  
Gauthier.

closed 2022-10-20 by fra589

MarkMassingill, 2023-02-20:
Hi @fra589 

I just wanted to say Thank You.  I've been doing a lot of reading today, trying to set up Mega 5x on a new CNC of my own design and the various issues you've addressed have so far covered all my needs.  The last on I read, the guy was a complete jerk to everyone and well deserved the black listing by you that he was awarded with.  I just wanted to let you know that for every one like him, I'm sure there are 100 who are as grateful as I am for your efforts to create Mega 5x in your own time on your own dime.  THANK YOU!

RaphaelDives, 2023-02-20:
👍

fra589, 2023-02-20:
Hi @MarkMassingill,  
  
You're welcome!
  
@++;
Gauthier.