fra589/grbl-Mega-5X #303: Homing dual axis

Host1992, 2022-10-16:
when trying to home with $HY both Y home fine, 
but when i'm trying with $H, cycle 1-2 work fine , cycle 3: only Y1 moves;

I tried a lot of parameter but I didnt find my problem can you help me @fra589 

[default.txt](https://github.com/fra589/grbl-Mega-5X/files/9795191/default.txt)
[config.txt](https://github.com/fra589/grbl-Mega-5X/files/9795186/config.txt)

closed 2022-10-16 by Host1992

fra589, 2022-10-20:
Hi @Host1992,  
  
When you use $HY, Grbl compute itself which axis needed to move, so, he move both Y axes.
When you use $H itself, it use the homing definition which is incorrect in your config.h.  
You should have:
`#define HOMING_CYCLE_2 ((1<<AXIS_2)|(1<<AXIS_4)) // Home Y axis`  
  
@++;  
Gauthier.

Host1992, 2022-10-20:
salut @fra589 ,
jai realiser ca apres m'avoir gratter le menton pendant 1h merci pour la reponse ! xD
jai ete oubliger de downgrade pour grbl 1.1h et mon vieux cncshield car apres avoir toute installer je me suis apercu que 5x ne supportais pas encore le square homing ;; 
j'ai bien essayer de trouver une solution mais mes connaissance en programmation sont loin d'etre suffisante donc je vais rester informer jai vu que cetait sur ton ''todo list'' ;