fra589/grbl-Mega-5X #304: GRBL Plotter Commands

jbaker1311, 2022-10-17:
Good afternoon, I'm trying to work out how to get my air assist working with grbl.
https://github.com/svenhb/GRBL-Plotter/issues/313
I've pinned them to D16 & D17 which need a signal to say on and off.
Previously I had them on coolant and mist but I needed them individually to turn off.
Now I'm working the commands to get them to work, any advice or answers?

Cheers Jacob

fra589, 2022-10-20:
Hi @jbaker1311,  
  
Standard pinout if D16 is digital output 0 and D17 is digital output 1.  

As described in http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65:  
- M62 P- - turn on digital output synchronized with motion. The P- word specifies the digital output number.
- M63 P- - turn off digital output synchronized with motion. The P- word specifies the digital output number.
- M64 P- - turn on digital output immediately. The P- word specifies the digital output number.
- M65 P- - turn off digital output immediately. The P- word specifies the digital output number.  

So, for example: to turn D16 ON, synchronized with motion, use the GCode command:  
`M62 P0`
to turn it of:  
`M63 P0`
  
@++;  
Gauthier.

jbaker1311, 2022-10-23:
Thanks for the swift reply, specifically I can't exactly get the commands to work with grbl plotter, not sure if you have any experience with that, I'm still waiting for a reply from Sven.

fra589, 2022-10-24:
Hi @jbaker1311,  
  
Is M62-M65 commands are working with a terminal (you can use [Putty](https://www.putty.org/) with serial connection at 115200 bps) or another GCode sender?  
Can you post the $I output to see which exact version of grbl-Mega-5X you have?  
  
@++;  
Gauthier.

jbaker1311, 2022-10-24:
Thanks for the reply, I've dumped the idea of using the digital outputs for now. As I can use flood and coolant to switch my relays. I haven't tried any other software as I'm only using my machine for a plotter/cutter and not a CNC/foam cutter. I can get back to you tomorrow with grbl version. Cheers

fra589, 2022-11-21:
No news from near a month... Issue closed.