fra589/grbl-Mega-5X #305: servo 0-5V

alexc57, 2022-11-14:
Hi, I am trying to use your firmware. OK for X et Y ( I don't have any Z).
My mother board is a mks base 1.6.
I have try to change value for spindle and attribute special pin for servo or pin for Z endstop as PWM. but I don't be able to found a solution.
I have found on github the pin out : MKS Base v1.6_004 SCH.
for X and Y I m agree but when attributed the pin for the servo (0-5V), nothing works. can you help me please. thks in advanced and best regards

fra589, 2022-11-21:
Hi @alexc57,  
  
I introduced since version 1.2g.20220917 the possibility of using an additional PWM output (GCode M67/M68 commands). Feature can be enabled with the `#define USE_OUTPUT_PWM` directive in config.h.  
  
The activation and the choice of the output pin is done in config.h, the PWM parameters are in cpu_map.h and the code is in analog_control.c.  
  
By adjusting the PWM parameters in cpu_map.h to set (among other things) the frequency compatible with the servo, it should be possible to drive it.  
https://www.google.com/search?q=atmega2560+pwm+frequency  
https://www.google.com/search?q=atmega2560+pwm+servo  
  
@++;  
Gautier.

alexc57, 2022-11-25:
thanks for the answer and details, trying many thing but nothing works.

fra589, 2022-11-27:
Duplicate entry with #209

closed 2022-11-27 by fra589