fra589/grbl-Mega-5X #310: Will not compile, homing cycles not defined.

ozzyman, 2022-12-28:
I have been bashing my head against this issue for days!

Issue:
Cannot compile. Following errors are raised:

In file included from /home/tony/Arduino/libraries/grbl/config.h:33:0,
                 from /home/tony/Projects/Plasma table/Software/grbl-Mega-5X-edge/grbl/examples/grblUpload/grblUpload.ino:1:
/home/tony/Arduino/libraries/grbl/grbl.h:72:4: error: #error "Required HOMING_CYCLE_0 not defined."
   #error "Required HOMING_CYCLE_0 not defined."
    ^~~~~
/home/tony/Arduino/libraries/grbl/grbl.h:100:4: error: #error "WCO refresh must be greater than one."
   #error "WCO refresh must be greater than one."
    ^~~~~
/home/tony/Arduino/libraries/grbl/grbl.h:103:4: error: #error "Override refresh must be greater than zero."
   #error "Override refresh must be greater than zero."
    ^~~~~

exit status 1

Compilation error: exit status 1

Issue appears to be that grbl.h is included BEFORE the definitions in config.h. 

Commenting out line 33 (#include "grbl.h" // For Arduino IDE compatibility.) results in clean compile.

fra589, 2022-12-29:
Hi @ozzyman,  
  
It seems to me to be the classic error of conflict between the Grbl files present in the Arduino library directories and others loaded in the Arduino IDE.
To make your configuration changes, you must edit the files separately and redo the compilation procedure at the beginning by deleting the old version present in the library directory.
cf. https://github.com/fra589/grbl-Mega-5X/wiki/Compiling-grbl-Mega-5X
[image]
  
When you compile Grbl, you should only have grblUpload.ino loaded in the Arduino IDE with no other files loaded.  
  
@++;
Gauthier.

fra589, 2023-01-08:
Hi @ozzyman,  
  
I don't know what happen on your system, but I just download the last Arduino IDE v2.0 appimage from Github (https://github.com/arduino/arduino-ide/releases/tag/2.0.3) and it work fine for me to compiling grbl-Mega-5X without any change to it.
See this video I made for this compiling test:
https://youtu.be/eFWJ6sa7IUk

@++;
Gauthier.

closed 2024-06-30 by fra589