fra589/grbl-Mega-5X #311: want to remap z axis step and direction pi

Ramr93, 2023-01-04:
hello sir . how can i remap ramps 1.6  z-axis step/direction pin to E1 port. haw can i edit it in cup_map.h.

RaphaelDives, 2023-01-04:
Hey,
You just have to switch the _PORT, _PIN and _DISABLED values for both axis…
I think Z is normally 2 and E1 should be 3.
It’s quite in the beginning of CPU_MAP.
Don’t forget about limit switches if you need them…

I‘m not shure if it is possible but you can‘t you just change the axis name in config.h? (without warranty…)

Cheers 
Raphael

fra589, 2023-01-04:
Hi @Ramr93, hi @RaphaelDives,  
  
Sure, the simplest answer is to give the name of 'Z' to the axis number 5. But this will impose to define 5 axes even if only 3 are connected and used...
in config.h:
```
#define N_AXIS 5
#define N_AXIS_LINEAR 5
...
#define AXIS_5_NAME 'Z'
```

Or do it by swapping axis ports 2 and 4 (in cpu_map.h, axis ports are numbered from 0 to 5).
in cpu_map.h:
```
#define STEP_PORT_2 C
#define STEP_PORT_4 L
#define STEP_BIT_2 1
#define STEP_BIT_4 3
#define DIRECTION_PORT_2 C
#define DIRECTION_PORT_4 L
#define DIRECTION_BIT_2 3
#define DIRECTION_BIT_4 1
#define STEPPER_DISABLE_PORT_2 C
#define STEPPER_DISABLE_PORT_4 K
#define STEPPER_DISABLE_BIT_2 7
#define STEPPER_DISABLE_BIT_4 0
```

@++;
Gauthier.

RaphaelDives, 2023-01-04:
Oh sorry… had the wrong numbers in my answer… 😕

Ramr93, 2023-01-05:
> Hey, You just have to switch the _PORT, _PIN and _DISABLED values for both axis… I think Z is normally 2 and E1 should be 3. It’s quite in the beginning of CPU_MAP. Don’t forget about limit switches [...]

Thank you for your response. I tried your way.  It not give any error But it doesn’t work.

Ramr93, 2023-01-05:
> Hi @Ramr93, hi @RaphaelDives,
>
> Sure, the simplest answer is to give the name of 'Z' to the axis number 5. But this will impose to define 5 axes even if only 3 are connected and used... in config.h: [...]

Thank you for your response. Will try and get back to you.

Ramr93, 2023-01-05:
> Hi @Ramr93, hi @RaphaelDives,
>
> Sure, the simplest answer is to give the name of 'Z' to the axis number 5. But this will impose to define 5 axes even if only 3 are connected and used... in config.h: [...]

Hi thank for your help.  It worked perfectly.  Thanks 😊 🙏🏾

closed 2023-01-05 by Ramr93