fra589/grbl-Mega-5X #313: Question for #define N_AXIS_LINEAR

ariefadha, 2023-01-11:
in config.h I have to state :
`#define N_AXIS_LINEAR 3 // Number of linears axis, must be <= N_AXIS
`
my question is, why grbl need to define linear axis? Is that related to limit switch for homing or because gcode linear translate to distance and rotational axis translate to degree radiant?

What if I state `#define N_AXIS_LINEAR 1` while the machine it self using 3 linear. And adjust the degree radiant by $101 and $102 to match the movement distance in actual movement?

RaphaelDives, 2023-01-11:
As I learned recently, this is just something about internal calculations and does only matter, if You use inches as measurement?!
(IF I recall that right - can't find it anymore...)
Had it wrong on my MPCNC for a long time myself.
Cheers Raphael

fra589, 2023-01-12:
Hi @ariefadha, Hi @RaphaelDives,  
  
Raphael, you are right 😃. N_AXIS_LINEAR is only used in G20 modal mode to use inches for length units (http://linuxcnc.org/docs/stable/html/gcode/g-code.html#gcode:g20-g21).  
  
In G21 modal mode (millimeters mode), Grbl assumes that the axis units are in millimeters and uses the coordinates of the movement orders without modifying them. In this mode, N_AXIS_LINEAR is never used.
When the G20 mode is used, all dimensions of linear axis movements are multiplied by 25.4 to ensure the conversion.  
This is achieved by loops that looks like this:  
```
for (idx=0; idx<N_AXIS_LINEAR; idx++) {
   if (bit_istrue(axis_dwords,bit(idx)) ) {
     gc_block.values.xyz[idx] *= MM_PER_INCH;
   }
}
```  
Anyway, you can use any other unit (radians for example) for any axes by staying in G21 mode and adjust the step per unit for those axes (parameter $100 to $105).  
You can even set N_AXIS_LINEAR = 0, this will completely disable G20 mode.
  
@++;  
Gauthier.

closed 2023-01-12 by fra589