fra589/grbl-Mega-5X #318: question about rotary

ariefadha, 2023-02-05:
Hi there,

I want my rotary axis only (not linear axis) ignore limit switch when working, and use limit switch for homing only. Is there any setup for that?

RaphaelDives, 2023-02-05:
Hey,
as long as you don‘t use hard limits, the limit switch is only working while homing.
Cheers
Raphael

ariefadha, 2023-02-05:
> Hey, as long as you don‘t use hard limits, the limit switch is only working while homing. Cheers Raphael

wait, so the soft limit not protect while working? i tot it limit the working area according value that we set according to axis length.

so I gues I need on/off switch for my rotary limit to disconnect while working, and connect while homming

RaphaelDives, 2023-02-05:
Yes of course they will protect your machine if enabled!
The question was about the limit switch, right?
On my Mpcnc they are both disabled. And my rotary homes normally with the limit switch and ignores it otherwise. (I wrote some homing A operations in my gcode while milling and start with A=0 again.)

I don’t know if there is some special behaviour for rotary axis. But maybe it’s possible to disable limits for special axis?! I don’t think it’s possible in standard config though… 😕

ariefadha, 2023-02-06:
wait, I got confused.
hard limit is using intrupt right? so whenever cnc move if we push the limit switch the machine will stop right?

I kinda read somewhere here on the issue page, on prev grbl machine only look hard limit when homming not when working. now the last release ver it use intrupt.

so if I use hard limit on rotary, it will stop rotating once it hit the hard limit (1 turn only) but you said "On my Mpcnc they are both disabled. And my rotary homes normally with the limit switch and ignores it otherwise."
is that mean you disable soft limit and enable hard limit, and it dont stop the rotational pass the limit? or it mean you disable both hard limit and soft limit via firmware and when homming grbl still detect the limit switch?

RaphaelDives, 2023-02-06:
Hey,
I understand it all the same! 
Hard limits use interrupts to stop the machine when triggered. Soft limits use internal calculation to check wether the next move will go over the defined limits. 
Bad written here https://github.com/fra589/grbl-Mega-5X/issues/140 
and in the Wiki https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration#130-131-132-133-134-135--xyzabc-max-travel-mm-degree
 you can disable soft limits for one axis by setting max travel to 0. I pretty shure that this only affects the soft limits and your hard limits will trigger anyway.
While homing, it does not matter if interrupts (hard limits) are enabled or not!
They will always work!

So the best solution for you should be: carefully check your soft limits (and set max travel to 0 for the rotary) and then disable hard limits. At least when you are using your rotary. 

And yes, I have soft and hardlimits disabled. But in my case the only thing happening when going over the limits is a loud noise from my steppers, so I can live with that! 😉

fra589, 2023-02-06:
Hi @ariefadha, hi @RaphaelDives,  
  
The homing algorithm use it's own process for finding limits switches independently of hard limit or soft limit functions.  
  
By default, the use of hard limits is disabled in the firmware (ENABLE_RAMPS_HW_LIMITS define is commented in cpu_map.h (line  number 206)) and so, the $21 (Hard limits) parameter is inoperative.
As on Arduino Mega + RAMPS limits pins are not all compatible with hardware interrupt, when ENABLE_RAMPS_HW_LIMITS is activated in the firmware, the hardware limit check code is added to the stepper interrupt code. This reduces performance a bit, so, I don't recommend using it.  
  
With good limit travels definitions ($130 to $135), the software limit calculation is enough to secure your machine movements.
Finally, you can use 0 for the limit travel of one particular axis (your rotary axis) if you want to disable limit check for this particular axis. The others will continue to be checked.  
  
@++;  
Gauthier.

ariefadha, 2023-02-06:
Hi @fra589 
So in short if I use a default (hard limit is disable), limit switch is ignore while my cnc (rotaty) in works. And the grbl only look the limit switch when in homming procedure? 

thats what I need, thanks

fra589, 2023-02-08:
Hi @ariefadha,  
  
Yes, this true.
Caution, if the software limits are active, do not forget to set the rotary axis max travel parameter to zero.  
  
@++;
Gauthier.

closed 2023-02-08 by fra589