fra589/grbl-Mega-5X #320: $27 - Homing pull-off, mm

ariefadha, 2023-02-09:
Is there any setting that allow to individually set the pull-off after homing using different values for each motor,  so the machine can squaring the axis (independent stepper on axis) by compensate for misaligned switches?

RaphaelDives, 2023-02-09:
We had this before…
https://github.com/fra589/grbl-Mega-5X/issues/29
Cheers Raphael

ariefadha, 2023-02-09:
> We had this before… #29 Cheers Raphael

so the answer is not? must to use mechanical approach for this?

too bad, this feature is very useful for 4 axis foam cutter cnc. which has independent axis.

fra589, 2023-02-09:
Hi @ariefadha, hi @RaphaelDives,  
  
Logical squaring of cloned axis is in the short todo list.
Development has started, but I can't announce a deadline... Be patient!
  
@++;  
Gauthier.

ariefadha, 2023-02-09:
> Hi @ariefadha, hi @RaphaelDives,
> 
> Logical squaring of cloned axis is in the short todo list. Development has started, but I can't announce a deadline... Be patient!
> 
> @++; Gauthier.

hi Gauthier thanks for your hard works. In the meantime, I think I can solved the squering to my machine. I'll attempt to make macro that do homming after that move the stepper that I need to adjust. as it only need once while the machine first time being on, I happy with that

closed 2023-02-09 by ariefadha