fra589/grbl-Mega-5X #321: Reset and lost parameters with six axes

jjnicholls, 2023-02-25:
I am working in a six axis machine, the movements must have fastest as possible, the main functions works well in lower feed and less microsteps, but in high feed rates (ie +4000) with eight or more microsteps it fail, in some cases, the parameters in EEPROM are overwrited to number 2147483.648, it seems like a overflow.

Have you any idea if there are limitations on six axis? if yes, what are? and, if is possible fix this in a config files or source code?    

My basics parameters are:

Grbl 1.2h ['$' for help]
$$
$0 = 10    (Step pulse time, microseconds)
$1 = 254    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.020    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 250.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 12000    (Maximum spindle speed, RPM)
$31 = 550    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 140.000    (X-axis travel resolution, step/mm)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 800.000    (Z-axis travel resolution, step/mm)
$103 = 140.000   
$104 = 140.000   
$105 = 140.000   
$110 = 4500.000    (X-axis maximum rate, mm/min)
$111 = 4500.000    (Y-axis maximum rate, mm/min)
$112 = 4500.000    (Z-axis maximum rate, mm/min)
$113 = 2500.000   
$114 = 2500.000   
$115 = 2500.000   
$120 = 50.000    (X-axis acceleration, mm/sec^2)
$121 = 50.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$123 = 50.000   
$124 = 50.000   
$125 = 50.000   
$130 = 1500.000    (X-axis maximum travel, millimeters)
$131 = 1500.000    (Y-axis maximum travel, millimeters)
$132 = 1500.000    (Z-axis maximum travel, millimeters)
$133 = 360.000   
$134 = 180.000   
$135 = 180.000 

This is an example of this fail when execute the command:

 $J=G21G91Y100F4500
ok
Grbl 1.2h ['$' for help]
$$
$0 = 117    (Step pulse time, microseconds)
$1 = 4    (Step idle delay, milliseconds)
$2 = 4    (Step pulse invert, mask)
$3 = 155    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 156    (Status report options, mask)
$11 = 0.000    (Junction deviation, millimeters)
$12 = 2147483.648    (Arc tolerance, millimeters)
$13 = 1    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 4    (Homing direction invert, mask)
$24 = 0.000    (Homing locate feed rate, mm/min)
$25 = 0.000    (Homing search seek rate, mm/min)
$26 = 4    (Homing switch debounce delay, milliseconds)
$27 = 0.000    (Homing switch pull-off distance, millimeters)
$30 = 32    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 0.000    (X-axis travel resolution, step/mm)
$101 = 0.000    (Y-axis travel resolution, step/mm)
$102 = 0.000    (Z-axis travel resolution, step/mm)
$103 = 0.000   
$104 = 0.000   
$105 = 0.000   
$110 = 2147483.648    (X-axis maximum rate, mm/min)
$111 = -0.000    (Y-axis maximum rate, mm/min)
$112 = 0.000    (Z-axis maximum rate, mm/min)
$113 = 2147483.648   
$114 = 32.000   
$115 = 0.000   
$120 = 0.000    (X-axis acceleration, mm/sec^2)
$121 = 0.000    (Y-axis acceleration, mm/sec^2)
$122 = -2147483.648    (Z-axis acceleration, mm/sec^2)
$123 = 0.000   
$124 = 0.000   
$125 = 0.000   
$130 = 0.000    (X-axis maximum travel, millimeters)
$131 = -0.000    (Y-axis maximum travel, millimeters)
$132 = -0.000    (Z-axis maximum travel, millimeters)
$133 = -0.000   
$134 = -0.000   
$135 = -0.000   
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0.0 S0]
ok

When is executed the command, It was reset and all parameters values change, (see $12 or $122 ). To continue I have to set parameters again. 

This is the screenshoot:

[image]

fra589, 2023-02-25:
Hi @jjnicholls,  

This question comes up quite often...

The stepper motors pulse rate needed to perform the movement you 're asking for seems way too high and is beyond the processing capabilities of the Atmega2560@8MHz processor.
Indeed, a simple calculation of the necessary frequency gives:
f = speed * step_per_mm / seconds_per_minutes
f = 4500 * 800 / 60 = 60000 Hz, which is more than 2 times too big!

To start, I advise you to read this discussion about the maximum frequency at which Grbl can go: https://github.com/grbl/grbl/issues/41
This is for the original version of Grbl on Atmega328p, but the Atmega2560 has the same architecture and frequency, so the limitations are pretty much the same.

You can then search the many discussions that have already taken place on this subject with the keywords "grbl max frequency" in Google: https://www.google.com/search?client=firefox-b-d&q=grbl+max +frequency

You will have to make a choice between the resolution of the microsteps and the speed. In general, speed and accuracy do not mix!
According to your parameters, the theoretical resolution is 1.25 micrometers... Is the mechanical construction of your machine so precise?
  
By searching a little more in Google (https://www.google.com/search?client=firefox-b-d&q=stepper+motor+microstepping+accuracy) we realize that microstepping does not bring so much precision than what we hope.

To reach the speed you want, you will have to reduce the microsteps or review the mechanics to reduce the number of steps per millimeter.

@++;  
Gauthier.

jjnicholls, 2023-03-01:
Hi, thank you for your reply.

I've been reviewing the information given, I was hoping this had already been improved in some way.

The need to have more microsteps is because in my machine (it's not exactly a CNC) high speed is needed with smooth acceleration without vibrations, it's not for precision.

I already tested with GRBLHAL using the LPC1769 driver and it works perfectly but it doesn't support more than 5 axes, and I haven't found a hardware compatible with GRBLHAL that has been tested with 6 axes. So my best option so far is grbl-Mega-5X.

Based on what you explain to me, I will continue testing until I get the best result.

One more thing, for your knowledge, the latest version of
grbl-Mega-5X does not connect with the latest versions of UGS (12,13,14), I have to use version 11, I don't know if it is a problem with UGS or grbl-Mega-5X

Thanks again.