fra589/grbl-Mega-5X #327: Can't connect to grblgru software

ariefadha, 2023-04-25:
I can't connecting to grblgru with my own compilation hex file but works fine with hex given by grblgru. heres the picture that showing the error :
[image]

might you guys can help me what I need to change before compiling

walterwissmann, 2023-04-25:
I tested an compiled grbl5x on Linux Mint serveral times the last days an tested it with 5D and bcnc...

No problems.

Walter

fra589, 2023-04-25:
Hi @ariefadha,  
I use several own compilation of grbl-Mega-5X without problem...
Do you have changed the "`GRBL_VERSION`" definition in grbl.h or the "`report_init_message()`" function in report.c?
  
Perhaps the problem come from your Grblgru, maybe you can ask to @GrblGru...

@++;
Gauthier.

ariefadha, 2023-04-25:
> Hey @ariefadha come on... some more information could be very useful:
> 
> How did you compile yout "own" hex file? (Does not sound like the suggested way via Arduino IDE...) Can you connect with other software/terminal? Did you try cn5x? (I like to use pronterface for testing...) Just guessing: Did you set the right baud rate? I think there was a change some time ago with the default value. (...)
> 
> Anyway it's very hard to help, with so less information...
> 
> Cheers, Raphael

other software works fine, terminal works fine. I use rckeith hex also has some problem.

and this is how i compile https://youtu.be/kIB8FFu5RLA

and it seems other users have some problem see comments section from this video https://youtu.be/jNKa2-UwKgo
[image]

ariefadha, 2023-04-25:
> I tested and compiled grbl5x on Linux Mint serveral times the last days and tested it with nc5x, bcnc...
> 
> No problems.
> 
> Walter

i never use cn5x. since my machine is 4 axis independent foam cutter with rotary table

GrblGru, 2023-04-25:
@ariefadha 

When establishing the USB connection I expect a normal ready message from Mega-5X. See pic below.

Which Mega-5X version are you using ?

When installing GrblGru, 2 Mega-5X bin files are copied into the installation directory. 
C:\Program Files (x86)\toe\GrblGru

Please try to flash the file  
grbl-Mega-5X-v1.2d_XYZU_SPINDLE_PWM_ON_D6.Hex
into the controller. In this file the 4-axes X,Y,Z and U are already selected, which you need for the operation of your FoamCutter.

wish you much success

[image]

ariefadha, 2023-04-25:
> @ariefadha
>
> When establishing the USB connection I expect a normal ready message from Mega-5X. See pic below.
>
> Which Mega-5X version are you using ?
>
> When installing GrblGru, 2 Mega-5X bin files are [...]

ive test grbl-Mega-5X-v1.2d_XYZU_SPINDLE_PWM_ON_D6.Hex file. it works fine. but i need 5 axis for rotary table and some rewiring limit switches.

GrblGru, 2023-04-25:
Ok, then try the other file as well.
grbl-Mega-5X-v1.2d_XYZAB_SPINDLE_PWM_ON_D6.hex.
It is configured for 5 axes.  But for 3 linear axes X,Y,Z and the two rotation axes A and B.
But you need 4 linear axes XYZU and one rotation axis A. Then you have to create your own version by changing the configuration file.
However, GrblGru does not have a corresponding FoamCutter model at the moment. But it is on top of my list.

ariefadha, 2023-04-25:
this is the ready message i got on serial terminal
[image]

and your current 4 axis model is oke-ish for me. but if you also add rotary table it will be awesome.

in the meantime i need to sovle the problem first how to connect to grblgru software, as you can see here in this video i need to set my own ver https://youtu.be/kIB8FFu5RLA

ariefadha, 2023-04-25:
> Then you have to create your own version by changing the configuration file

can you elaborate what setting must i change so it can connect? because with lastest ver and all default, i still cant connect

GrblGru, 2023-04-25:
I am very sure that Gauthier can tell you much better how to configure Mega-5X.
But I noticed that in your 1st screenshot there is the message "Waiting for Grbl identification". This only appears when you try to connect to GRBL. But you have to select "GRBL-Mega-5X" in the selection window.
Then the message "Waiting for Mega-5X identification" appears.

[image]

ariefadha, 2023-04-25:
> I am very sure that Gauthier can tell you much better how to configure Mega-5X. But I noticed that in your 1st screenshot there is the message "Waiting for Grbl identification". This only appears [...]

oh, i forgot to tell you that im using grblgru v4.8. because above that version it got bug on displaying the machine. here i give you screenshot using both, as you can see on v5.0 the machine axis not draw good.

ver4.8:
[image]

ver5.0:
[image]

and some other bug like slow animation when i change viewing angle, and uninstalling the software just like deleting desktop shortcut, still not delete the installation folder, and i can still run that.

ariefadha, 2023-04-25:
> Hi @ariefadha,
> 
> Your compiling method and config seams to be good... The ready message you got on serial terminal is the standard one.
> 
> But it seams that Grblgru's version is adding the axes names in parenthesis... @GrblGru, do you have changed the "`GRBL_VERSION`" definition in grbl.h or the "`report_init_message()`" function in report.c?
> 
> You can try with my own Grbl driver software cn5X (https://github.com/fra589/cn5X). it automatically adapts to all configurations from 3 to 6 axes, linear or rotary. There is no drawing of the part or the machine, but it analyzes and displays all the status information of grbl-Mega-5X and allows to drive all its functions, including the one I added against to the original Grbl.
> 
> [image]
> 
> @++; Gauthier.

thanks, I will try cn5x also.

GrblGru, 2023-04-26:
@Gauthier
Just for your information. I made 2 Mega-5X versions with different configuration, 
because many people who use GrblGru are pure users and not developers who can build their own version.
So I can see which version is used, I added the extra identifier in the version string. But it is not mandatory for the operation with GrblGru.

@ariefadha
Your screenshots show me that you used the very old version of the FoamCutter. But there is already a new better version. I recommend you to use the newest GrblGru version.

To learn how to use the FoamCutter, see the last video in the Video Gallery on my website.
https://www.grblgru.com/VideoGallery
The on-screen texts are unfortunately in German only, but the function should be clear.

Let me know if you have any other questions or suggestions for improvement.

GrblGru, 2023-04-26:
Oh, sorry. My mistake. I mean @fra589  , of course.
Copy& paste is not always correct :)

gauthier, 2023-04-26:
No problem ;) that way, the actual recipient will be kept in touch :)

ariefadha, 2023-04-26:
> 

i already install the lastest ver of grblgru, but that really bugging me is the 3d machine cannot move the gantry like you see on my 2nd image above. thats why i use oldest version. 

and can you tell me what setting you change when compile grbl mega? because my 1st concern here is to able to connect to grblgru 1st.

is changing GRBL_VERSION" definition in grbl.h or the "report_init_message()" will solve the problem?

GrblGru, 2023-04-26:
Ok, I think you didn't understand my last post, but let's first see why your version doesn't work with GrblGru.
Please provide me with your bin file here.
(Rename to .txt)

ariefadha, 2023-04-27:
> Ok, I think you didn't understand my last post, but let's first see why your version doesn't work with GrblGru. Please provide me with your bin file here. (Rename to .txt)

ok I'll test version 5.1.2 once I got home.

ariefadha, 2023-04-28:
@GrblGru all works well with ver 5.1.2 that I've tried. many thanks for the hard works.

I have some suggestions for improvement :
1. pendant/keyboard(shortcut) to jog foam cutter machine.
2. debug/command line bar at the bottom always hidden on program start.
3. "show current (yellow) toolpath" some how stuck while operating machine (tested in foam cutter machine only). but works fine in simulation mode.
4. uninstalling program does not take effect. it still not remove the file on program file > toe > grblgru .. and still can running the program after uninstallation.

but once again thanks for the hardworking, im having fun with your software, especially the simulation and macro programming that allows users have more flexibility. can't wait for your next ver. danke schoon

closed 2023-04-28 by ariefadha