fra589/grbl-Mega-5X #329: b-axis configuration

sfmissionmark, 2023-05-08:
I am very new to cnc so I am sure I am just holding something wrong here. I have a 4 axis cnc with a rotary running in the Y orientation.

I believe I configured everything correctly. Connecting with serial monitor in Arduino IDE allows me to send G0 B-1. G0 A-1 errors with error:20. I changed the settings in config.h to this. Admittedly I am guessing.

// Axis indexing and names
#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V, W, D, E & H.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'
#define AXIS_4 2
#define AXIS_4_NAME 'B'

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'B' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'A' // Letter of axis number 5
#endif
#if N_AXIS > 5
  #define AXIS_6 5
  #define AXIS_6_NAME 'C' // Letter of axis number 6
#endif
#if N_AXIS > 6
  #error "N_AXIS must be <= 6. N_AXIS > 6 is not implemented."
#endif

// Renaming axis doesn't change their number. By default, the status report give axis values in
// the order of their number. Some graphical interface are not able to affect axis values reported
// by Grbl to the correct axis name.
// Uncomment to enable sorting of axis values by axis_names rather than by axis number. Default disabled.
// If this option is enabled, the sorting order will be X, Y, Z, U, V, W, A, B, C, D, E & H 
// as defined below.
#define SORT_REPORT_BY_AXIS_NAME
//#define AXIS_NAME_SORT_ORDER {'X', 'Y', 'Z', 'U', 'V', 'W', 'A', 'B', 'C', 'D', 'E', 'H'}
#define AXIS_NAME_SORT_ORDER {'X', 'Y', 'Z', 'B'}

My question is none of the ui applications properly pick up the B axis. I tried ugs and openbuilds ui. These all display there is still an an axis. Of course trying to click the A axis in the ui fails with an error. Is there some other configuration I am missing or does the ui just not properly work with b axis?

sfmissionmark, 2023-05-09:
I realized the easy solution is to configure it as 5 axis and just disable the a-axis in the ui. I am still curious what I have wrong.

fra589, 2023-05-10:
Hi @sfmissionmark,  
  
Your 4 axes config is the right one, but... 

Your problems with UI are just UI developers problem who are not able to understand than may be other configs than what they have in their brains.  
When I started adding the axes to Grbl and created grbl-Mega-5X capable of just about every standard axes configuration possible, I quickly realized that none of the UIs would be able to handle them. That's why I also wrote my own UI: cn5X available here: https://github.com/fra589/cn5X  
  
@++;  
Gauthier.

sfmissionmark, 2023-12-15:
Turning off A axis makes this ui usable. Closing.

closed 2023-12-15 by sfmissionmark