fra589/grbl-Mega-5X #330: invert function of the limit switch

hamelsveld, 2023-05-09:
Hi,

I've limit switches installed on my machine. My switches are continuously on meaning the if I don’t hit my limit switch the loop is closed. 

If I put $5 to “0” they are all triggerd by default and will go to off when I hit the switch. <-- this tells me the switches are working and the software detects movement. However I want it to be "off" by default and "triggerd" when a switch is pushed. 

By reading the config it tells me that if I change $5 to "1" it should invert the switch if I'm correct. However if I do that they are all triggerd  by default and nothing happens if I hit the switch.

is there any way I can reverse the function of the switch?

Rc Keith pointed out some options around line 343 but not being a developer I'm not sure what option (if any) to use....

Please advice,

Gr Sven

fra589, 2023-05-10:
Hi @hamelsveld,  
  
As I wrote in the wiki (https://github.com/fra589/grbl-Mega-5X/wiki/Limit-switches-and-homing), using $5=1 mean that all axes defined limit switch (AXIS_1 min & AXIS_1 max, AXIS_2 min & AXIS_2 max, ..., AXIS_N min & AXIS_N max) must be normally closed switch.  
  
<< In the normally closed mode, all the switches being closed when idle. If a switch is open, the limit is trigered for the axis.
This effectively assumes that ALL switches are effectively closed when idle.
Warning! If one of the contactors is missing, the coresponding pin must be connected to the ground in order for the NF logic to be respected. Otherwise, the limit switch on the axis concerned will be trigered permanently. >>  
  
@++;  
Gauthier.

hamelsveld, 2023-05-10:
Hi Gauthier,

Thanks for your reply. One more question I’ve all the same switches and connected the X-Y-Z ones to the board and $5 set to “1”. Then they are all triggered and not responding.

Does this means I also need to connect all other pins on the board from “s” to “-“ before the function starts to work again?

Gr Sven

fra589, 2023-05-10:
@hamelsveld,  
  
Yes.
If your config is N_AXIS = 3 (X, Y & Z) and your 3 limits switches connected on Xmax, Ymax and Zmax on the RAMP board, you need to connect the Xmin, Ymin and Zmin signals pins to the ground.  
  
@++;  
Gauthier.

hamelsveld, 2023-05-10:
Great then final 2 questions

Where/what setting is N_AXIS ?  is that one of the $ options in my settings?

And do you know where the pins are on the RAMPS board?  Are this the ones next to the x-y-z I’m currently using?

Sorry to bother you so much

Greetings Sven

hamelsveld, 2023-05-10:
Hi,

With your help I managed to get the switch to work in the software and now when I touch the switch the trigger goes from “off” to “triggerd” Iike I wanted. But when I enable hard-limits ($21) it should according to the description in to alarm and stop. But If I flip a swich that does not happen.

Do I need to do more?

Greetings Sven

fra589, 2023-05-10:
Hi @hamelsveld,  
  
It's normal that when hard limits are enabled, Grbl will stop and switch to Alarm mode when any switch is triggered.  
  
In my point of view, enabling hardware limits is not the best method to deal with machine limits. When the hardware limit is triggered, the system will immediately stop, without ensuring compliance with the acceleration parameters. This result of loosing position and so, a new homing will be necessary.  
  
The best solution in my opinion, is to correctly put the right dimensions of al axes in parameters and use the soft limits enabled ($20=1).  
  
@++;
Gauthier.

fra589, 2023-05-10:
In reply to the precedents and new questions: 
N_AXIS is defined (like axis names) at compile time in the file config.h.
By default, the hardware limits are not enabled at compile time for performances reasons (checking hardware limits use few CUP cycles for each movement).
To enable it, you should uncomment the line 206 in ccpu_map.h: `#define ENABLE_RAMPS_HW_LIMITS`
  
For the pin attribution and location, read the wiki: https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout
  
@++;
Gauthier.

hamelsveld, 2023-05-10:
great thanks then we can close this issue / question

thanks again

closed 2023-05-10 by hamelsveld