fra589/grbl-Mega-5X #332: Questions about using 4 axis and 5 motos without the ramps shield.

rbp9802, 2023-05-29:
Hello fra589, thank you for making this incredible work!

I'm planning to make a 4 axis CNC machine (cloned Y axis) with 5xNema23 steppers and 5xTB6600 drivers and I have some questions before i start wiring all up. 

I'm trying to avoid the ramps shield since i have seen in some youtube videos that is it not necesary and, from what I undestood, it just make the wiring easier. So:

- Is it necessary to use the ramps shield in order to use the 5 motors? 
- And why does the cpu_map.h only accept 5 stepper motors with the ramps shield and 3 without it?
- Can i connect the pins directly to the arduino mega board like if I'm using the ramps shield (but actually not using it)? 
- What are the benefits of using the ramps shield?

Thanks in advance

fra589, 2023-05-30:
Hi @rbp9802,  

>     * Is it necessary to use the ramps shield in order to use the 5 motors?

The RAMPS shield is not necessary at all. It make the wiring easier and add necessary components (capacitor & jumpers) for A4988/DRV8825 or equivalents form factor drivers. But with TB6600 nothing of this is necessary.
The RAMPS also add the ability to use 12v PWM output to directly drive the spindle. Without RAMPS, PWM outputs are only 0-5v.

>     * And why does the cpu_map.h only accept 5 stepper motors with the ramps shield and 3 without it?

??? Which version of grbl-Mega-5X have you? Since 2021 April/May, 3 axes only old code have been suppressed and there is only the CPU_MAP_2560_RAMPS_BOARD definition in the cpu_map.h file

>     * Can i connect the pins directly to the arduino mega board like if I'm using the ramps shield (but actually not using it)?

Yes, sure! 

>     * What are the benefits of using the ramps shield?

See first reply... With it, you can use Grbl on standard 3D printer hardware without any changes.

> Thanks in advance

Your welcome!

> *Update: Stepper motors worked perfectly fine without ramps shield attached, checking limit switches still pending. I don't know if ramps shield have and internal capacitor that improve the perfomance of the switch, like stated on `line 480` of `config.h` file.

??? line 480 of config.h file doesn't mention capacitor:
`480   // By default, Grbl sets all input pins to normal-high operation with their internal pull-up resistors`
Are you sure you have the latest [grbl-Mega-5X](https://github.com/fra589/grbl-Mega-5X) version?
In anyway, The RAMPS pins limits switches are directly connected to the ATmega2560 pins without capacitor or filter.

@++;
Gauthier.

rbp9802, 2023-05-30:
Thank you for your reply Gauthier. 

`Which version of grbl-Mega-5X have you?`
Sorry, looks like I open opened another grbl library on my pc. I already delete it.

`line 480 of config.h file doesn't mention capacitor`
Sorry again, I meant the paragragh. It actually mentions low pass filter on line 482.
```
482 // although its recommended that users take the extra step of wiring in low-pass filter to reduce
483 // electrical noise detected by the pin.
```

`The RAMPS pins limits switches are directly connected to the ATmega2560 pins without capacitor or filter.`
Thank you for the information.

closed 2023-05-31 by fra589