fra589/grbl-Mega-5X #336: Another homing issue

Zjokkeh, 2023-06-05:
Hi, 

I am trying for several hours now to get the homing working without any luck. I have read all the previous issues and I can't find a solution.

I have a 5 axis machine with dual X and dual Y (mpcnc). The switches are NO and working fine. They show op perfectly in the status rapport.

The only thing I noticed is when I increase the pull-off distance $27 the axis will keep on going longer after the switches are reached. I looks like he doesn't pull-off but keeps on going. After a while I get an alarm 8 from grbl see attachment.
[image]

My version:

[16:17:59] [ $i ] [VER:1.2h.20220109:]

[16:17:59] [ $i ] [AXS:5:XYZXY]

[16:17:59] [ $i ] [OPT:VNMGH,35,255,24]

[16:17:59] [ $i ] ok

[grbl-settings-backup-2023-06-05-d.txt](https://github.com/fra589/grbl-Mega-5X/files/11653495/grbl-settings-backup-2023-06-05-d.txt)

[config.txt](https://github.com/fra589/grbl-Mega-5X/files/11653516/config.txt)

fra589, 2023-06-05:
Hi @Zjokkeh,  
  
It looks like a movement direction management problem... as if the output pin that defines the direction of movement for the stepper motor driver does not work...

Apart from the homing function, do the motors all work in both directions and without inversion?

Did you reset the flash memory with the ` $RST=*`  command after flashing grbl-Mega-5X in the arduino board?

@++;
Gauthier

RaphaelDives, 2023-06-05:
Hey,
another MPCNC out there!! :)

Do you have dual endstops, or just one per axis?
and are both motors going on or just one of them?
What happens, if you trigger them by hand?
Did you modify the pins in cpu_map?
Looks to me like the endstops are not correctly triggered, or not correctly recognized during homing cycle.
(MPCNC builds tend to run the limit wires parallel to the motor wires... so the switches seems to work correctly, but could fail when motors are on...)
At least - it can work! ;)
Cheers
Raphael

RaphaelDives, 2023-06-05:
Hey @Zjokkeh,
wow that's a very nice setup! :) Just out of interest: What size is it? Looks quite big on the picture...

I would try the following: Just for testing, connect two switches with a short cable directly to the Ramps board, and try if you can trigger them manually while homing.

On the first glance I didn't notice anything, but I will compare my config to your's when I'm at home.

Cheers
Raphael

RaphaelDives, 2023-06-05:
I'd say thats quite a strange behaviour for NO (=normally open) switches, since disconnecting a NO switch would just mean, that it is never triggered and homing would search for it until it reaches the max_travel value...

So with disconnected switches, homing should travel towards the switches and not the other way round!
(someone correct me, if I'm wrong...)

By the way... your pulloff distance is much too high. Some mm are quite enough! I'm alright with the standard value - 5mm or so...
Try it with a smaller value, and I guess the homing will do two short travels. and give an error then.
That would mean, your switches are constantly triggered while homing. And it never homes, but tries to clear the (triggered) switches, which will fail... Because of the long pulloff it LOOKS like homing, but it is not!

With disconnected (or short) switches, the homing direction has to be towards the switches!! (You have to adjust $23 for that)
With triggered switches the homing/pulloff move has to be the other direction. (You can just try that!)

Good luck! :)
Raphael

Zjokkeh, 2023-06-06:
I  did some late night testing, but with no result. Tested the switches with a multimeter and they work perfectly. Connected switches directly to the ramps board with no difference. And I forgot to mention that I first tested the system with the mpcnc marlin software for dual axis autosquare and it worked just fine then... Same setup only NC for the marlin...

fra589, 2023-06-06:
Hi @Zjokkeh,  
  
Does your limit switch are on the max side or on the min side ?

@++;
Gauthier.

Zjokkeh, 2023-06-06:
Tried both. Max, min, reverse, everything that I could think off...

RaphaelDives, 2023-06-06:
Hm, sounds quite difficult...
(Somehow strange that Marlin works. Maybe Marlin uses another debouncing strategy?!)
Since I'm using grbl-mega-5x on my MPCNC (with rotary axis), I'm very confident that can work on your machine as well.

I think your axis and homing definition in the config should be okay (Assuming you did not change anything else).
By the way, I think you have some errors in your settings. Axis dependent parameters have to be the same for cloned axis!! E.g. $100 (X1 axis) must be the same as $103 (X2 axis) it's the same for $110 and $113, $111 and $114 and so on... You should check your max feed rate, acceleration and max travel! And as said: $27 is much too high! You need just as much, that the limit switches are disengaged.

BUT I don't think these settings are causing your actual problems... The question is, how to nail down the problem.
And why does the limit switches not work...

Did you correct your homing direction? (Without switches connected it should drive in the direction towards the switch - both sides!!)
Then of course the config mentioned above...
As you said jogging X and Y works? (in the right direction?)
Then just to ask... how did you connect yout switches? One side to -, the other side to S? And yes, stupid question, but you connect them on the right ports, and they show up with right names? (X=axis1/XX=axis2 etc.)
What happens with homing when the switches are triggered?

Finally, I'm not getting tired to tell about interferences with limit switches escpecially when you run the wires next to the motor current. (But this was the test with the extra short cable switches...)

That's all I can think of at the moment. Maybe it is an idea to start new with a fresh installation of grbl-mega-5x in case you changed something different? ;)
 
Cheers Raphael

Zjokkeh, 2023-06-06:
I started with NC because I wanted to switch between marlin and grbl with minimum effort. But gave the same result.

Zjokkeh, 2023-06-06:
> This is not good because of your wrong Homing dir invert mask...
Correct the Homing dir invert mask ($23) and after, adjust the motors direction to have the good homing direction.

Do you mean by switching the coil wires from the steppers to adjust the motor direction?

fra589, 2023-06-06:
You can switch the coils wires or adjust the step direction invert mask ($3) to have motors running to the right homing direction...

> the jogging is switched around...

Perhaps it's the jogging command which is false...
What happen with the jogging command if you enable the soft limits ($20=1)?

Zjokkeh, 2023-06-06:
soft limits enabled gives ALARM 15 after some jogs.

fra589, 2023-06-06:
Clarification concerning the hypothesis of a wrong Jogging command: 
By default, Grbl defines machine space as all negative, as do most CNCs.
When homing, with limit switches on the max side (which is the default), the machine axis position is set to à (zero). With limit switches on the min side, the machine position is set to -max_travel(axis).
This mean that all movements in machine position must e in negatives coordinates. They have to have destination coordinates between 0 and -max_travel(axis).
To use movements with positives coordinates destination, you need to define the coordinate system of working position by defining the G92 and/or G53-G59 offset.

fra589, 2023-06-06:
> soft limits enabled gives ALARM 15 after some jogs.

ALARM 15 does not exist??? 
There is only 11 alarms defined in the Grbl code.
[ALARM 15](https://github.com/fra589/grbl-Mega-5X/blob/edge/doc/csv/alarm_codes_en_US.csv)

Zjokkeh, 2023-06-06:
Sorry I meant error 15
[image]

I am lost with your explanation...

fra589, 2023-06-06:
Yes, error: 15 is the normal behavior with destination movement outside of machine enveloppe.
If you haven't defined the working coordinate (G92 or G53 to G59), all movements should be in machine coordinates.
This mean that destination coordinates must be all negatives.

What is the output of the `?` command ?

Zjokkeh, 2023-06-06:
[19:06:17] [ ? ] <Idle|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>

[19:06:43] [ $# ] [PRB:0.000,0.000,0.000,0.000,0.000:0]

fra589, 2023-06-06:
> I am lost with your explanation...

You're not alone having difficulties with coordinates systems of milling CNC...
It's a fairly complex system that allows you to do a lot of things with the GCode...
This may be why CNC programming is a profession in its own right with a fairly long and complex training :-)

We will have to add this theme to the wiki...

> [19:06:17] [ ? ] <Idle|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>
> [19:06:43] [ $# ] [PRB:0.000,0.000,0.000,0.000,0.000:0]

Your `$#` output is not complete... But as I understand, it should be:
```[G54:-0.000,0.000,0.000,0.000,0.000]
[G55:0.000,0.000,0.000,0.000,0.000]
[G56:0.000,0.000,0.000,0.000,0.000]
[G57:0.000,0.000,0.000,0.000,0.000]
[G58:0.000,0.000,0.000,0.000,0.000]
[G59:0.000,0.000,0.000,0.000,0.000]
[G28:0.000,0.000,0.000,0.000,0.000]
[G30:0.000,0.000,0.000,0.000,0.000]
[G92:0.000,0.000,0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000,0.000,0.000:0]
ok
```

This mean what I say, there is no working coordinates systems defined in our machine. So, you can only use negative coordinates.

When Grbl is started, the positions of each axis is 0 (zero). Once an axis is homed, the the machine origin for that axis is set to the homed position: 0 if the limit switch is on the max side, -max_travel(axis) if the limit switch is on the min side.
So, the only destination movements of an axis can be between 0 and -max_travel(axis).

Example for your case without homing (assuming correct X axis max travel of 1000 mm):
- Disable homing cycle by issuing $22=0,
- Place the X axis near to it's maximum position, then reset or power on Grbl,
- The machine position will be 0 on all axes `<Idle|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>`: 
- Jog the X axis to it's middle position: `$J=G90G21X-500F700` (at this moment, G90 or G91 will do the same because the axis is at his origin position)
- Define the origin of the X axis at this point with the G92 global offset: `G92X0.0`
=> After that, the destination coordinates of the X axis movements must be between -500 mm and +500 mm.

The LinuxCNC documentation is a good place to see details of working coordinates system:
http://linuxcnc.org/docs/html/gcode/coordinates.html

@++;
Gauthier.

Zjokkeh, 2023-06-06:
So I cannot use the auto home function to the minimum side? I am lost here 😂

Zjokkeh, 2023-06-06:
[21:14:37] [ $# ] [G54:3.000,0.000,0.000,3.000,0.000]

[21:14:37] [ $# ] [G55:0.000,0.000,0.000,0.000,0.000]

[21:14:37] [ $# ] [G56:0.000,0.000,0.000,0.000,0.000]

[21:14:37] [ $# ] [G57:0.000,0.000,0.000,0.000,0.000]

[21:14:37] [ $# ] [G58:0.000,0.000,0.000,0.000,0.000]

[21:14:37] [ $# ] [G59:0.000,0.000,0.000,0.000,0.000]

[21:14:37] [ $# ] [G28:0.000,0.000,0.000,0.000,0.000]

[21:14:37] [ $# ] [G30:0.000,0.000,0.000,0.000,0.000]

[21:14:37] [ $# ] [G92:0.000,0.000,0.000,0.000,0.000]

[21:14:37] [ $# ] [TLO:0.000]

[21:14:37] [ [ PROBE ] ] Probe move ERROR - probe did not make contact within specified distance

[21:14:37] [ $# ] [PRB:0.000,0.000,0.000,0.000,0.000:0]

Zjokkeh, 2023-06-07:
Sorry for my ignorance. I am new to all of this. 

Every time I start the machine I should:
- move all axis to max position
- move all axis to middle position
- use G92 to set 0
- then home to min position... 

Is this correct? 

Why does marlin for mpcnc work without any fuss? Because it is altered especially for mpcnc?

Zjokkeh, 2023-06-07:
>Hey,
I don't know what's the correct way, but I
start my machine,
home it and then
jog to my starting point and
set the zero for my needs (and z height... don't have a tool length sensor yet)
and start milling...
(I'm using bcnc on my raspberry btw)

This is what I want. Why does this not work with my setup???

fra589, 2023-06-07:
No, it was just an example to illustrate grbl coordinate system (and more generally CNC coordinates systems) with G92 origin positioning.

http://linuxcnc.org/docs/html/gcode/coordinates.html
Have you read the link I mentioned above? Read it and re-read it carrefully.

Search Google for "cnc coordinate system" and "grbl coordinate system" and "G53 G54-G59 G92 GCode" to have many more info about this.

It is essential to understand this to be able to use a CNC milling machine...

RaphaelDives, 2023-06-07:
> Tell me in childs language. Why won't it home?

That is the million dollar question. I set up grbl several times on three different machines. 
But that was never a problem (despite of my interferences on my MPCNC, that caused similar problems to yours... Sometimes it worked, sometimes not and most of the time only when the homing travel was not to far and it was very random which motors stopped, sometimes one, sometimes both... 🤔)

Zjokkeh, 2023-06-07:
> [12:32:10] [ $$ ] $0=10 ;Step pulse time, microseconds

[12:32:10] [ $$ ] $1=254 ;Step idle delay, milliseconds

[12:32:10] [ $$ ] $2=0 ;Step pulse invert, mask

[12:32:10] [ $$ ] $3=0 ;Step direction invert, mask

[12:32:10] [ $$ ] $4=0 ;Invert step enable pin, boolean

[12:32:10] [ $$ ] $5=0 ;Invert limit pins, boolean/mask

[12:32:10] [ $$ ] $6=0 ;Invert probe pin, boolean

[12:32:10] [ $$ ] $10=0 ;Status report options, mask

[12:32:10] [ $$ ] $11=0.020 ;Junction deviation, millimeters

[12:32:10] [ $$ ] $12=0.002 ;Arc tolerance, millimeters

[12:32:10] [ $$ ] $13=0 ;Report in inches, boolean

[12:32:10] [ $$ ] $20=1 ;Soft limits enable, boolean

[12:32:10] [ $$ ] $21=0 ;Hard limits enable, boolean

[12:32:10] [ $$ ] $22=1 ;Homing cycle enable, boolean (Grbl) / mask (GrblHAL)

[12:32:10] [ $$ ] $23=0 ;Homing direction invert, mask

[12:32:10] [ $$ ] $24=25.000 ;Homing locate feed rate, mm/min

[12:32:10] [ $$ ] $25=250.000 ;Homing search seek rate, mm/min

[12:32:10] [ $$ ] $26=250 ;Homing switch debounce delay, milliseconds

[12:32:10] [ $$ ] $27=5.000 ;Homing switch pull-off distance, millimeters

[12:32:10] [ $$ ] $30=12000 ;Maximum spindle speed, RPM

[12:32:10] [ $$ ] $31=550 ;Minimum spindle speed, RPM

[12:32:10] [ $$ ] $32=0 ;Laser-mode enable, boolean

[12:32:10] [ $$ ] $100=100.000 ;X-axis steps per millimeter

[12:32:10] [ $$ ] $101=100.000 ;Y-axis steps per millimeter

[12:32:10] [ $$ ] $102=100.000 ;Z-axis steps per millimeter

[12:32:10] [ $$ ] $103=100.000 ;A-axis steps per degree

[12:32:10] [ $$ ] $104=100.000 ;unknown

[12:32:10] [ $$ ] $110=750.000 ;X-axis maximum rate, mm/min

[12:32:10] [ $$ ] $111=750.000 ;Y-axis maximum rate, mm/min

[12:32:10] [ $$ ] $112=750.000 ;Z-axis maximum rate, mm/min

[12:32:10] [ $$ ] $113=750.000 ;A-axis maximum rate, deg/min

[12:32:10] [ $$ ] $114=750.000 ;unknown

[12:32:10] [ $$ ] $120=50.000 ;X-axis acceleration, mm/sec^2

[12:32:10] [ $$ ] $121=50.000 ;Y-axis acceleration, mm/sec^2

[12:32:10] [ $$ ] $122=50.000 ;Z-axis acceleration, mm/sec^2

[12:32:10] [ $$ ] $123=50.000 ;A-axis acceleration, deg/sec^2

[12:32:10] [ $$ ] $124=50.000 ;unknown

[12:32:10] [ $$ ] $130=1200.000 ;X-axis maximum travel, millimeters

[12:32:10] [ $$ ] $131=1200.000 ;Y-axis maximum travel, millimeters

[12:32:10] [ $$ ] $132=1200.000 ;Z-axis maximum travel, millimeters

[12:32:10] [ $$ ] $133=1200.000 ;A-axis maximum travel, degrees

[12:32:10] [ $$ ] $134=1200.000 ;unknown

These are my settings for this moment. If I want to jog left or right for 0.1mm I get the error 15 that I exceed the machine travel. How is this possible? My max travel is about 1200 mm so my settings are right, not?

Zjokkeh, 2023-06-07:
I believe its noise on the cables to the switches. If I just something longer than a 20cm dupont wire to the switches grbl thinks the switches are closed. Is there a solution for this?

RaphaelDives, 2023-06-07:
That was my intention with
> I would try the following: Just for testing, connect two switches with a short cable directly to the Ramps board, and try if you can trigger them manually while homing.

and

> Finally, I'm not getting tired to tell about interferences with limit switches escpecially when you run the wires next to the motor current. (But this was the test with the extra short cable switches...)

I always had this interference problem - now, I'm running 24v through my switches.
I connect 24v on one side and Ground to the other side and the output goes through a 24v relay board (just search on aliexpress - I think I have the 8 channel version), and the output of the relais go to the arduino as a "normal open switch".
Not the most elegant solution for shure, but it's working rock solid and the relay board is really not expensive...
(I also added a z-max sensor: https://www.thingiverse.com/thing:6010255 ;) )

Not a very satisfying solution, I know... 
And I still don't get why marlin can do the job.

Cheers Raphael

RaphaelDives, 2023-06-07:
> I believe its noise on the cables to the switches. If I use something longer than a 20cm dupont wire to the switches grbl thinks the switches are closed. Is there a solution for this?

ah yes... I tried everything, with external pullups and noise gates /filters and... and....
Relays did the job finally. :D

Zjokkeh, 2023-06-07:
> Not a very satisfying solution, I know...
And I still don't get why marlin can do the job.

The creator of marlin must have encounterd this problem also and perhaps he found a superb solution for the noise problem.

I will try the relay board way, I have some laying around I should look for it.

Anyway I am so glad I found this problem, and the solution was right there in the beginning. Thanks a lot for helping me out.

Perhaps an idea to put this noise issue in the wiki?

fra589, 2023-06-07:
Here is the Grbl's machine coordinate:

[image]

_(I talk only about X and Y to simplify explanations, but it the same behavior with all axis)_

1) Internally, Grbl know only negative coordinates. The soft limits control only if the machine coordinate destination is between -maxTravel and 0,
2) At power on, before homing, the machines coordinates are initialized to 0 (zero) whatever it's real position.
If you don't define any other coordinate system (G92 or G54 to G59), the only moves you can do is to negative destinations:
- `$J=G91G21X10F700` will not work and will throw an error 15 because Grbl's machine coordinates destination is not between -max_travel and 0 (it can't be positive).
- `$J=G91G21X-10F700` will work, and will move the X axis by 5mm in the Xmin direction. **do it before testing homing and verify that it move in the right (X-) direction. If no, inverse the motor direction by switching the coil wires from the steppers**.

Do the same for Y in negative direction:
- `$J=G91G21Y-10F700` **do it before testing homing and verify that it move in the right (Y-) direction. If no, inverse the motor direction by switching the coil wires from the steppers**.

At power on or reset, Grbl can't know the real position of axes, so they are initialized to 0 (zero) whatever their real positions. To make the true limits of the machine coincide with those of Grbl, homing is necessary.
If homing to the positive direction, when the limit switch of an axis is triggered, it's coordinate is initialized to 0 (zero). This is the default, but it's not what you want...

> I want my homing bottom left corner. So X- and Y-.

You say that you want to have homing at min side (I assume also z-)...
**So, you need to define the homing dir invert mask: `$23=31`**
If you don't do this, homing will not work because the homing logical direction will not be the correct one.

When homing to the negative direction, when the limit switch of an axis is triggered, it's coordinate is initialized to -max_travel (negative value). This mean that soft limits will be the same for Grbl: between -max_travel and 0.

To test the homing configuration independently of a possible hardware problem, I advise you to shortly connect simple normally open switches between limits pin and the ground. Then:
- Issue the `$H` homing command,
- verify that the axis goes to the good direction,
- manually trig the switch before the end of the movement, the axis should stop and goes to the opposite direction for homing switch pull-off distance ($27=5.000 in your case),
- release the switch before the end of the 5mm movement, the axis should stop and goes to the homing direction,
- manually trig the switch again, the axis will stop it's movement, this is the end of homing for this axis.

If it manually work and it don't work with your reals switches on your machine, it will be the proof of hardware problem.

@++;
Gauthier.

Zjokkeh, 2023-06-08:
With the relay board it works indeed. Just a small question: shouldn't the steppers keep being powered after the homing/squaring to keep them in position? Just like in marlin?

RaphaelDives, 2023-06-08:
https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration#1---step-idle-delay-milliseconds

Zjokkeh, 2023-06-08:
> Yes you should do that…https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration#1---step-idle-delay-milliseconds

Tnx man, I should learn how to read...

closed 2023-06-09 by fra589