fra589/grbl-Mega-5X #339: homing limit switches not stopping motors

markusca, 2023-06-26:
I recently revisited connecting NO limit switches to my 5 axis cnc. 
Currently I seem to get proper feedback in the ui and cli that a limit switch was hit. 
But none of the motors stop when trying to run a homing cycle or jogging and the limit switch is hit. 
I think I have read everything dozens of time and I cant find anything I am missing.

(?) With all limit switches not triggered.
<Idle|MPos:0.0000,0.0000,0.0000,0.0000,0.0000|Bf:35,254|FS:0.0,0|WCO:0.0000,0.0000,0.0000,0.0000,0.0000>

With x+ limit switch being triggered.
<Idle|MPos:0.0000,0.0000,0.0000,0.0000,0.0000|Bf:35,254|FS:0.0,0|Pn:X|Ov:100,100,100>

$0=10
$1=254
$2=0
$3=16
$4=0
$5=0
$6=0
$10=19
$11=0.020
$12=0.002
$13=1
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=250.000
$26=250
$27=5.000
$30=20000
$31=550
$32=0
$100=320.397
$101=320.667
$102=320.000
$103=11.050
$104=10.020
$110=1999.800
$111=1999.800
$112=1500.000
$113=30000.000
$114=1440.000
$120=50.000
$121=50.000
$122=50.000
$123=1000.000
$124=50.000
$130=400.000
$131=200.000
$132=200.000
$133=360.000
$134=180.000

Attached my config file.
[config.zip](https://github.com/fra589/grbl-Mega-5X/files/11861332/config.zip)

RaphaelDives, 2023-06-26:
Hey,
just some quick thoughts… Did you try the limit switches with motors engaged or disabled? And how did you connect them? Did you think about pull ups (or downs)? And proper filtering? (My standard question: do the limit switch wired run parallel to the motor wires?)

Then, we recently had a problem with NO switches: https://github.com/fra589/grbl-Mega-5X/issues/336
Maybe this could help…

Cheers
Raphael

fra589, 2023-06-26:
Hi @markusca,  
  
Reading your config.h file, there is an error with your axis names and definition:
[image]
=> You added first AXIS_4 definition which overwrites the AXIS_3 definition... You need to correct it.

To use hardware limits ($21=1 in your configuration), you need to enable it at compile time by uncomment the line 206 in the cpu_map.h file:
[image]

But at the risk of repeating what I have already written several times, it seems to me preferable and more efficient to use the software limits with a good configuration of the dimensions of the machine.  
  
@++;
Gauthier.

markusca, 2023-07-03:
I don't think I realized the homing process was separate from the limit switch process. It makes sense that using hard limits to control the limits is fine if your homing process works correctly.

I started completely over with a new checkout because I had guessed so many times I really didn't know what I had. The only modification I made to config was to comment out HOMING_CYCLE_0 and change HOMING_CYCLE_1 to HOMING_CYCLE_0. I don't have limit switches on z.

Current grbl values is this:
 
Grbl 1.2h ['$' for help]
[MSG:'$H'|'$X' to unlock]
$0=10
$1=254
$2=0
$3=16
$4=0
$5=0
$6=0
$10=19
$11=0.020
$12=0.002
$13=1
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=250.000
$26=250
$27=5.000
$30=20000
$31=550
$32=0
$100=320.397
$101=320.667
$102=320.000
$103=11.050
$104=10.020
$110=1999.800
$111=1999.800
$112=1500.000
$113=30000.000
$114=1440.000
$120=50.000
$121=50.000
$122=50.000
$123=1000.000
$124=50.000
$130=400.000
$131=200.000
$132=200.000
$133=360.000
$134=180.000
ok

I am only testing in Arduino serial connection to avoid any other issues with things like UGS hiding output. I only have one limit switch connected NO on the max side of the x axis all others are disconnected.

If I send G0 X1 the axis moves toward the right side of the bed.
If I send $HX the X axis moves 2 times toward the left side of the bed and then stops.
If I send $HX and interrupt the process by manually closing the limit switch I get the following.
ALARM:8
ok

Grbl 1.2h ['$' for help]
[MSG:'$H'|'$X' to unlock]

I really appreciate your help. Could you tell me what I might be doing wrong?

RaphaelDives, 2023-07-04:
Hey @markusca,
when I remember correctly, moving two times while homing seems like grbl tries to disengage the limit switch. (Had that on my z axis with no limit switch installed…)
Did you test the correct function of your limit switches? 
Then I will give the same advice over and over again… disconnect your limit switches from ramps, connect switches with very short cable and trigger them manually while homing.
In this case I’m pretty shure you have interferences on your switches… (let me guess: you’re running the wires parallel to the motor wires…? 😉)
Even if your switches would work correctly while testing, they could be randomly triggers while operating your motors.
So long story short… use the same voltage for the switches as for motors and put Relais to them (or anything similar…).

Cheers Raphael

fra589, 2023-07-04:
hi @markusca,  
  
Yes, it's right than the homing code is independent than the hard limit process.  
  

> I started completely over with a new checkout because I had guessed so many times I really didn't know what I had. The only modification I made to config was to comment out HOMING_CYCLE_0 and change HOMING_CYCLE_1 to HOMING_CYCLE_0. I don't have limit switches on z.
(...)
I am only testing in Arduino serial connection to avoid any other issues with things like UGS hiding output. I only have one limit switch connected NO on the max side of the x axis all others are disconnected.  
  
If I have a good understand of that you made, in a fresh install of grbl-Mega-5X, you only change the homing cycle in the config.h file. So, you should have:  
- #define N_AXIS 5
- #define N_AXIS_LINEAR 3
- #define AXIS_1_NAME 'X'
- #define AXIS_2_NAME 'Y'
- #define AXIS_3_NAME 'Z'
- #define AXIS_4_NAME 'A'
- #define AXIS_5_NAME 'B'
- #define HOMING_CYCLE_0 ((1<<AXIS_1)|(1<<AXIS_2))

active in your config.h file.
  
Here is a description of the homing algorithm:
- When sending the <$HX> command, the X axis moves a first time in the positive direction (if the homing direction invert mask (parameter $23) doesn't invert it), with a maximum distance equal to the "X-axis maximum travel" parameter ($130) at "Homing locate feed rate" speed parameter ($24),
  - If the limit switch is not triggered before moving at max travel parameter of the axis, the alarm 9 is triggered,
- when the limit switch of the axis is triggered first, the X axis will move in the negative direction for a distance of "Homing switch pull-off distance" parameter ($27) to disengage the limit switch,
  - If the limit switch is not disengaged withing this distance, an alarm 8 is triggered,
- when the pull-off distance is reached with the limit switch cleared, the X axis will move a second time in the positive direction at "Homing search seek rate" speed parameter ($25). The $25 parameter should be slower than $24 to to determine its location accurately.
  - If the limit switch is not triggered before moving at max travel parameter of the axis, the alarm 9 is triggered,
- when the limit switch of the axis is triggered this second time, the X axis will move in the negative direction for a distance of "Homing switch pull-off distance" parameter ($27) to disengage the limit switch, then it stop.
- The homing sequence is finished for the X axis and it's machine coordinate is initialized to "- Homing switch pull-off distance" parameter (negative value of $27).

When you issue the $HX command, the axis should change it's direction when you manually close the switch, manually open the switch a soon it change it's direction, then, the axis should change it's direction another time, manually close the switch a second time and open it again, the homing process will be finished.
If the axis detect the switch and does not change direction, you probably have a hardware problem with the pin direction of it.
If the axis doesn't detect the limit switch, you probably have a switch problem.

@++;
Gauthier.

sfmissionmark, 2023-07-04:
Thank you Raphael and Gauthier. I am an idiot. I carefully loomed my switches with shielded wire on the machine. Then inside the control box I neatly wrapped non shielded cable along side the high speed stepper wires. Of course it did exactly as the software was written. Immediately triggering the switch and then attempting to back off twice and failing to un-trigger the switch. Of course that all works wonderful when the steppers are not running. Again I appreciate all the help and hopefully that is the last of my stupid questions.

SlavaStels, 2023-08-04:
Hello @fra589 !

First of all thank you for your work!

I also have an issue with switch triggering while homing. I am not a SW/FW engineer so I have no clue (yet) whether I have a HW or a FW issue. 

Setup:
I have a 4-Axis (XYZU all independent) cutting machine built following Keith RC manuals. I have Arduino Mega 2560 + RAMPS 1.4 running "grbl-Mega-5X-firmware-1.02" . I have soft limits with 4 physical switches that are NC. I use GRBL HotWire Mega 5X - v5.13 to run the machine.

Issue:
Switches can be triggered manually (simply press them) when machine is powered, but motors hold position (not in homing mode). However, when I start homing procedure (starts with Z-axis) none of the switches can be triggered. When I push them during homing "Limit Switch Status" does not change.

I use a single unshielded cable with 12 wires for one side (2 steppers + 2 switches), so I thought it might be the noise issue or something. Therefore, I decided to connect Z-axis limit switch to RAMPS separately via cable which is only 50mm long to eliminate the noise probability, but it still does not want to work. 

Is it possible that there is something wrong with part of FW that describes the homing sequence (maybe pins are not mapped correctly or something)?

Please let me know if you need more specific information (part of config.h or other directory etc).

Thank you.

fra589, 2023-08-04:
Hi @SlavaStels,  
  
By default, Grbl is designed to use Normally-Open (NO) switches. You want to use Normally-Closed (NC) switches, the parameter $5 must be fixed to 1 (`$5=1`) .  
  
When you do this, all unused limits pins must be connected to the ground to avoid Grbl interprets this open pin as triggered.  
  
@++;
Gauthier.

SlavaStels, 2023-08-04:
> Hi @SlavaStels,
>
> By default, Grbl is designed to use Normally-Open (NO) switches. You want to use Normally-Closed (NC) switches, the parameter $5 must be fixed to 1 (`$5=1`) .
>
> When you do this, all [...]

Thank you for the fast reply, I did not expect that!)

I have $5=1 and $20=1 and switches work as intended (triggered only when pressed) when motors are in standby mode (motors are ON but not rotating). When I run homing cycle none of the switches can be triggered.

I have min limit switches on D3, D14, D18 and D42 pins. Switches connected between Signal(S) and negative(-).

fra589, 2023-08-05:
Hi @SlavaStels,  
  
> triggered only when pressed

How do you verify than Grbl show the switch triggered ?

The homing process have his own detection of triggering limits switch, different than other limits process. And then, there is no report of switch detection during the homing process.
During the homing, both max limit and min limit of an axis are interpreted same by the code, and then, with $5=1, your max limites are always open, so viewed as triggered.
In your case, you need to short circuit to the ground the pins: D2, D15, D19 and D40.

@++;
Gauthier.

SlavaStels, 2023-08-05:
Hi @fra589,

Under setting tab in Grbl HotWire Mega 5X - v5.13 there is a "Limit Switch Status" pane with four boxes named X,Y,Z and U. At the moment by default there is no tick sign when switches are not triggered, but when switches are triggered tick sign show up in a corresponding box. It works perfectly fine when machine in a standby mode (tick appears where it should when I trigger switches) but is does not appear when I am running Homing Cycle.

[image]

My limit switches are wired like on the picture:

[image]

I actually decided to check the video from Keith RC on limit switches once again and realized that "Limit Switch Status" does not show online status of the switches when in Homing Cycle, but for Keith everything works as intended (Homing Cycle is successful).

[https://youtu.be/T2yhzRf1N0w?t=3101](url)

Could it be that I messed up some settings related to feeds and acceleration?

I have a belt driven machine (no lead screws), so I had to play with step/mm, speed and acceleration to calibrate the machine and make it move faster. 

Here is my preset:

ID Value Description 
$0=10 step pulse, usec
$1=255 step idle delay, msec
$2=0 step port invert mask:
$3=13 dir port invert mask:
$4=0 step enable invert, bool
$5=1 limit pins invert, bool
$6=0 Not Used
$10=3 status report mask
$11=0.020 junction deviation, mm
$12=0.002 arc tolerance, mm
$13=0 report inches, bool
$20=1 soft limits, bool
$21=0 hard limits, bool
$22=1 homing cycle, bool
$23=1 homing dir invert mask
$24=200.000 homing feed, mm/min
$25=100.000 homing seek, mm/min
$26=5 homing debounce, msec
$27=10.000 homing pull-off, mm
$30=1000 Voltage Divisor Max
$31=0 Not Used
$32=0 Not Used
$100=40.000 x, step/mm
$101=40.000 y, step/mm
$102=40.000 z, step/mm
$103=40.000 u, step/mm
$110=2400.000 x max rate, mm/min
$111=2400.000 y max rate, mm/min
$112=2400.000 z max rate, mm/min
$113=2400.000 u  max rate, mm/min
$120=1000.000 x accel, mm/sec^2
$121=1000.000 y accel, mm/sec^2
$122=1000.000 z accel, mm/sec^2
$123=1000.000 u  accel, mm/sec^2
$130=500.000 x max travel, mm
$131=350.000 y max travel, mm
$132=350.000 z max travel, mm
$133=500.000 u max travel, mm

fra589, 2023-08-06:
Hi @SlavaStels,  
  
Can you post your full config.h and cpu_map.h files ?

@++;
Gauthier.

SlavaStels, 2023-08-06:
Hi @fra589,

Sure, here they are :

[config and cpu_map.zip](https://github.com/fra589/grbl-Mega-5X/files/12269876/config.and.cpu_map.zip)

Here is also the FW I use, I took config and cpu_map from it:

[grbl-Mega-5X-firmware-1.02.zip](https://github.com/fra589/grbl-Mega-5X/files/12269877/grbl-Mega-5X-firmware-1.02.zip)

I hope that helps.

closed 2024-06-30 by fra589