fra589/grbl-Mega-5X #342: Changing homing order

DonArkon, 2023-07-30:
I want to change the order in which tie axis home. I need to do this since the endstops are stationary at the x/u motor mount. Therefore they need to be homed first so the y/z gantry can reach their endstops.

I went to config.h and tinkered with:
`#ifdef DEFAULTS_RAMPS_BOARD
  #if N_AXIS == 4 // 4 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_1) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_1 (1<<AXIS_2) // Home 4th axis (A)
    #define HOMING_CYCLE_2 (1<<AXIS_3) // Home X axis
    #define HOMING_CYCLE_3 (1<<AXIS_4) // Home Y axis
  #elif N_AXIS == 5 // 5 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_4) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_1 (1<<AXIS_3) // Home 4th axis (A)
    #define HOMING_CYCLE_2 (1<<AXIS_5) // Home 5th axis (B)
    #define HOMING_CYCLE_3 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_4 (1<<AXIS_2) // Home Y axis
  #elif N_AXIS == 6 // 6 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_4) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_1 (1<<AXIS_3) // Home 4th axis (A)
    #define HOMING_CYCLE_2 (1<<AXIS_5) // Home 5th axis (B)
    #define HOMING_CYCLE_3 (1<<AXIS_6) // Home 6th axis (C)
    #define HOMING_CYCLE_4 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_5 (1<<AXIS_2) // Home Y axis
  #else // Classic 3 axis
    #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_1 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_2 (1<<AXIS_2) // Home Y axis
  #endif
#else
  #define HOMING_CYCLE_0 (1<<AXIS_3)                // REQUIRED: First move Z to clear workspace.
  #define HOMING_CYCLE_1 ((1<<AXIS_1)|(1<<AXIS_2))  // OPTIONAL: Then move X,Y at the same time.
  // #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
#endif // DEFAULTS_RAMPS_BOARD`

I switched the AXIS_? around but it did not change the order axis homed after reflashing. As control software I'm using GRBL HotWire Mega 5X V5.13

Is the change in config.h supposed to be enough or is the homing order also effected by the windows application?

fra589, 2023-07-30:
Hi @DonArkon,   

Yes, adjust homing in config.h is enough to made that you want... 
But, to understand what appens for your particular configuration, the section which defining N_AXIS and AXIS_*_NAME is missing in your post.

@++;
Gauthier.

DonArkon, 2023-07-31:
Thx for your reply.

I haven't changed that part, that's why I did not include it into my post :)
Another missing info might be the boards I'm using. It's a Keyeestudio §D MKS Gen V1.4
https://www.keyestudio.com/products/keyestudio-3d-mks-gen-v14-printer-motherboard-control-board-for-arduino-3d-printer
But from what I see it's just a "fancy" RAMPS 1.4 and should work without further adjustments.

```
// To use with RAMPS 1.4 Board, comment out the above defines and uncomment the next two defines
#define DEFAULTS_RAMPS_BOARD
#define CPU_MAP_2560_RAMPS_BOARD

// Serial baud rate
// #define BAUD_RATE 230400
#define BAUD_RATE 115200

// Axis array index values. Must start with 0 and be continuous.
#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  //#define N_AXIS 5            // Number of axes
  //#define N_AXIS_LINEAR 3     // Number of linears axis
  #define N_AXIS  4          // Number of axes KH
  #define N_AXIS_LINEAR 4     // Number of linears axis KH
#else
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V & W.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  //#define AXIS_4_NAME 'A' // Letter of axis number 4
  #define AXIS_4_NAME 'U' // Letter of axis number 4 KH
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'V' // Letter of axis number 5
#endif
#if N_AXIS > 5
  #define AXIS_6 5
  #define AXIS_6_NAME 'W' // Letter of axis number 6
#endif
#if N_AXIS > 6
  #error "N_AXIS must be <= 6. N_AXIS > 6 is not implemented."
#endif
```

fra589, 2023-07-31:
Hi @DonArkon,  
  
OK, I see that your N_AXIS is 4 (`#define N_AXIS  4`)...

So, you need to adjust the homing definition in the right section (between `#if N_AXIS == 4` and `#elif N_AXIS == 5`): 

```
#if N_AXIS == 4 // 4 axis : homing
#define HOMING_CYCLE_0 (1<<AXIS_1) // Home 1st axis (X)
#define HOMING_CYCLE_1 (1<<AXIS_4) // Home 4th axis (U)
#define HOMING_CYCLE_2 (1<<AXIS_3) // Home 3rd axis (Z)
#define HOMING_CYCLE_3 (1<<AXIS_2) // Home 2bd axis (Y)
#elif N_AXIS == 5 // 5 axis : homing
```

With this definition, the homing order will be: AXIS_1, AXIS_4, AXIS_3 and AXIS_2 (X, U, Z and Y).

The important things are `#define HOMING_CYCLE_<cycle number> (1<<AXIS_<axis number>)`
With <cycle number> starting at 0 and has an increment of 1,
And <axis number> from 1 to N_AXIS.
The rest of the lines after `//` are comments which are not compiled in the firmware. 

@++;  
Gauthier.

DonArkon, 2023-08-05:
Got it working...

The embarrising solution? You have to modify the correct config.h file....

closed 2023-08-06 by fra589