fra589/grbl-Mega-5X #343: Cann't use  Servo motor MG90S 180degree

akkarawat2000, 2023-09-21:
Hi, Now I have problem about Servo motor MG90S 180degree  5V

I cann't use with grbl-Mega-5X  I try to connect with Pin D6 on Ramp board Version 1.5  but not working and 
I try to connect direct on Arduino Mega2560  on Pin D6  it's not working too. 

and I try to change config.h  like this

//----------------------------------------------------------------------
// Spindle, laser and other PWM output
//----------------------------------------------------------------------
// Chose the spindle pin output :
// SPINDLE_PWM_ON_D8  => 0-12v 16 bits PWM on RAMPS D8 (default)
// SPINDLE_PWM_ON_D9  => 0-12v 8 bits PWM on RAMPS D9
// SPINDLE_PWM_ON_D6  => 0-5v 8bits PWM on RAMPS Servo 2 signal (Mega 2560 D6)
// Uncomment the line which correspond to your hardware
//#define SPINDLE_PWM_ON_D8
#define SPINDLE_PWM_ON_D6
//#define SPINDLE_PWM_ON_D9

// Spindle PWM signal inversion:
// In case of particular electronics, it may be necessary to invert the values
// of the PWM signal of the spindle. For example, if the minimum spindle 
// rpm is 1 and maximum is 1000, M3S250 will output 75% instead of 25% and
// M3S750 will output 25% instead of 75%. Disabled by default
//#define INVERT_SPINDLE_PWM_VALUES

// Use different spindle output pin in laser mode:
// Spindle or laser tools do not have the same hardware specifications.
// When using both spindle and laser on the same machine it will be useful
// to have spindle and laser on diffrents pins which can deliver the
// differents outputs nedded.
//----------------------------------------------------------------------
// ! IMPORTANT: When changing the SEPARATE_SPINDLE_LASER_PIN compil option,
// don't forget to issue the reset factory defaults Grbl command: $RST=*
// if you forget the $RST=* command after change, Grbl may have 
// unpredictable behavior!
//----------------------------------------------------------------------
// Uncomment the next line to enable this functionality (default disabled):
//#define SEPARATE_SPINDLE_LASER_PIN

#ifdef SEPARATE_SPINDLE_LASER_PIN
  // Laser PWM can be on D6 (default) or on D8 or D9. 
  #define LASER_PWM_ON_D6
  //#define LASER_PWM_ON_D8
  //#define LASER_PWM_ON_D9
#endif

// Use output PWM drived by GCode command M67(Analog Output,Synchronized) 
// or GCode command M68(Analog Output, Immediate).
//----------------------------------------------------------------------
// ! IMPORTANT: When changing the USE_OUTPUT_PWM compil option,
// don't forget to issue the reset factory defaults Grbl command: $RST=*
// if you forget the $RST=* command after change, Grbl may have 
// unpredictable behavior!
//----------------------------------------------------------------------
// Uncomment the next line to enable the use of M67/M68 PWM output (Disabled by default).
#define USE_OUTPUT_PWM

#ifdef USE_OUTPUT_PWM
  // Optional PWM can be on D9 (default) or on D8 or D6. 
  // Warning ! Optional can't use the same timer than the spindle or the laser pin
  // For more information about this, see the comment of the relevant section in cpu_map.h 
//  #define OUTPUT_PWM_ON_D9
  //#define OUTPUT_PWM_ON_D8
  #define OUTPUT_PWM_ON_D6
#endif

-----------------------------

but not working if try to use with command M3 S255   and M4 255 

Command response ok but Motor not rotate or working  I connect with GND and 5V external too   but 

I try to test another way by write simple source code control Servo motor  and i connect Pin D6,D9,D11 on Arduino Mega 2560 board 
with 5V External or 5V internal of Arduino Mega2560  it's working fine  but if use Pin D6 on Ramp Board D6  not working now.

I would like to know how to solve this problem ? 
I think this issue  very important . and if possible please make video how to setup 
or how to use grbl-Mega-5X with Servo motor on Youtube .

 I think it's easy to people understand too. 

Because Beginner like me not understand all from your document . 
your document some information not have detail too much , not explain how to use step by step, it's hard for me understand. 

-------------------
2 . on D10 and D9 working fine now .  
 i don't know if use D10 or D9 use with servo motor 5V or not ? 
May be you have trick.

I hope you understand me and help me this problem too. 

Thank you very much.

akkarawat2000, 2023-09-21:
cpumap.h    i do like this 

 // Define spindle enable and spindle direction output pins.
  #define SPINDLE_ENABLE_DDR      DDRG
  #define SPINDLE_ENABLE_PORT     PORTG
  #define SPINDLE_ENABLE_BIT      5 // MEGA2560 Digital Pin 4 - Ramps 1.4 Servo 4 Signal pin (D4)
  #define SPINDLE_DIRECTION_DDR   DDRE
  #define SPINDLE_DIRECTION_PORT  PORTE
  #define SPINDLE_DIRECTION_BIT   3 // MEGA2560 Digital Pin 5 - Ramps 1.4 Servo 3 Signal pin (D5)

  // Define flood and mist coolant enable output pins.
  #define COOLANT_FLOOD_DDR   DDRB
  #define COOLANT_FLOOD_PORT  PORTB
  #define COOLANT_FLOOD_BIT   4 // MEGA2560 Digital Pin 10 - Ramps 1.4 12v output
  #define COOLANT_MIST_DDR    DDRH
  #define COOLANT_MIST_PORT   PORTH
  #define COOLANT_MIST_BIT    6 // MEGA2560 Digital Pin 9 - Ramps 1.4 12v output

  #if defined(SPINDLE_PWM_ON_D8)

    // Set Timer up to use TIMER4B which is attached to Digital Pin 8 - Ramps 1.4 12v output with heat sink
    #define SPINDLE_PWM_MAX_VALUE     1024.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
    #ifndef SPINDLE_PWM_MIN_VALUE
      #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
    #endif
    #define SPINDLE_PWM_OFF_VALUE     0
    #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

    //Control Digital Pin 8
    #define SPINDLE_TCCRA_REGISTER    TCCR4A
    #define SPINDLE_TCCRB_REGISTER    TCCR4B
    #define SPINDLE_OCR_REGISTER      OCR4C
    #define SPINDLE_COMB_BIT          COM4C1

    // 1/8 Prescaler, 16-bit Fast PWM mode
    #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
    #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
    #define SPINDLE_OCRA_REGISTER   OCR4A // 16-bit Fast PWM mode requires top reset value stored here.
    #define SPINDLE_OCRA_TOP_VALUE  0x400 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

    // Define spindle output pins.
    #define SPINDLE_PWM_DDR   DDRH
    #define SPINDLE_PWM_PORT  PORTH
    #define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 8

  #elif defined (SPINDLE_PWM_ON_D6)

    // Set Timer up to use TIMER4C which is attached to Digital Pin 6 - Ramps Servo 2
    #define SPINDLE_PWM_MAX_VALUE     255.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
    #ifndef SPINDLE_PWM_MIN_VALUE
      #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
    #endif
    #define SPINDLE_PWM_OFF_VALUE     0
    #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

    //Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
    #define SPINDLE_TCCRA_REGISTER    TCCR4A
    #define SPINDLE_TCCRB_REGISTER    TCCR4B
    #define SPINDLE_OCR_REGISTER      OCR4A
    #define SPINDLE_COMB_BIT          COM4A1

    // 1/8 Prescaler, 8-bit Fast PWM mode
    #define SPINDLE_TCCRA_INIT_MASK (1<<WGM41)
    #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
    #define SPINDLE_OCRA_REGISTER   ICR4 // 8-bit Fast PWM mode requires top reset value stored here.
    #define SPINDLE_OCRA_TOP_VALUE  0xFF // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

    // Define spindle output pins.
    #define SPINDLE_PWM_DDR   DDRH
    #define SPINDLE_PWM_PORT  PORTH
[... 207 lines cut ...]

      //Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
      #define OUTPUT_TCCRA_REGISTER    TCCR4A
      #define OUTPUT_TCCRB_REGISTER    TCCR4B
      #define OUTPUT_OCR_REGISTER      OCR4A
      #define OUTPUT_COMB_BIT          COM4A1

      // 1/8 Prescaler, 8-bit Fast PWM mode
      #define OUTPUT_TCCRA_INIT_MASK (1<<WGM41)
      #define OUTPUT_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
      #define OUTPUT_OCRA_REGISTER   ICR4 // 8-bit Fast PWM mode requires top reset value stored here.
      #define OUTPUT_OCRA_TOP_VALUE  0xFF // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

      // Define spindle output pins.
      #define OUTPUT_PWM_DDR   DDRH
      #define OUTPUT_PWM_PORT  PORTH
      #define OUTPUT_PWM_BIT   3 // MEGA2560 Digital Pin 6

    #else
      #error "OUTPUT_PWM_ON_D9 or OUTPUT_PWM_ON_D8 or OUTPUT_PWM_ON_D6 must be defined in config.h with the USE_OUTPUT_PWM option!"
    #endif

  #endif // USE_OUTPUT_PWM

#endif // CPU_MAP_2560_RAMPS_BOARD

fra589, 2023-09-21:
Hi @akkarawat2000,  
  
Servos need a specific PWM frequency to operate. This is not currently supported by grblMega-5X, but it's in the TODO list...

@++;
Gauthier
P.S. I close this issue since it's duplicate from the #209 one.

closed 2023-09-21 by fra589

akkarawat2000, 2023-09-21:
When you will make this feature finish ?  I see last question in year 2022.
This possible can do in this year 2023 ? 
I think this feature control servo motor very important because I must use with Arm Robot 5 Axis now .
I think your firmware GRBL now is the best if you have this feature it's very good 
many people will can use many application too. 

I hope you can do it soon 

or it's possible if i write source code in Arduino IDE for control Servo motor by my self and add to source code your GRBL ? 
I don't know how to add it. if you know please suggest too. 

Thank you

svenhb, 2023-09-21:
Check #323, I already mentioned the needed changes

akkarawat2000, 2023-09-22:
> Check #323, I already mentioned the needed changes

Yes , but fra589 still not edit and add this feature for use easy now right ?  and 

I see your comment this 

This means, following changes are needed in cpu_map.h:
line 333 #define SPINDLE_PWM_MAX_VALUE     512.0 // Translates to about 30 Hz PWM frequency at 1/1024 prescaler
line 348 #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS40) | (1<<CS42))
line 350 #define SPINDLE_OCRA_TOP_VALUE  0x200 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

you test ready ? and work or not ? 

Thank you.

svenhb, 2023-09-22:
These changes I would try (they are similar to the Arduino Uno changes for Servo use).
But I never tried, because I don't need it yet. I test my software with a bare Arduino Duo without any connected hardware.

svenhb, 2023-09-22:
You know that in fact the spindle PWM is used for servo, and not the ports from the RAMPs board?

akkarawat2000, 2023-10-10:
Hi , Today I test servo motor MG90S 180degree 5V again 
I change like you suggest below 

This means, following changes are needed in cpu_map.h:
line 333 #define SPINDLE_PWM_MAX_VALUE     512.0 // Translates to about 30 Hz PWM frequency at 1/1024 prescaler
line 348 #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS40) | (1<<CS42))
line 350 #define SPINDLE_OCRA_TOP_VALUE  0x200 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

and compile with upload 

I connect servo motor at RampBoard 1.5  at   D6 , 5V ,GND 

and I send Gcode 
M3 S36
ok
M3 S10
ok 

but don't have notthing happend  , Servo motor not rotate  .

How should i fixed this problem ? 

Thank you

svenhb, 2023-10-10:
Are you sure you edit the correct files? After following this: https://github.com/fra589/grbl-Mega-5X/wiki/Compiling-grbl-Mega-5X The files are copied to the user/documents folder... 
I checked correct files by editing also #define GRBL_VERSION_BUILD "20220109" in grbl.h

akkarawat2000, 2023-10-10:
Yes , I sure use #define GRBL_VERSION_BUILD "20220109"  this version

https://github.com/fra589/grbl-Mega-5X/tree/edge/grbl

you test like me and working now ?

What's servor you use ? and you connect D6 Channel ?

svenhb, 2023-10-10:
I think it was D8 because of line 128 in config.h #define SPINDLE_PWM_ON_D8

akkarawat2000, 2023-10-10:
D8 can connect Servo 5V ? 

I think D8 support just 12V  please confirm me.

akkarawat2000, 2023-10-10:
[image]

svenhb, 2023-10-10:
[image]

akkarawat2000, 2023-10-10:
Now I use like this right ?   
I use like this last time i comment to you  but not work 

//----------------------------------------------------------------------
// Spindle, laser and other PWM output
//----------------------------------------------------------------------
// Chose the spindle pin output :
// SPINDLE_PWM_ON_D8  => 0-12v 16 bits PWM on RAMPS D8 (default)
// SPINDLE_PWM_ON_D9  => 0-12v 8 bits PWM on RAMPS D9
// SPINDLE_PWM_ON_D6  => 0-5v 8bits PWM on RAMPS Servo 2 signal (Mega 2560 D6)
// Uncomment the line which correspond to your hardware
//#define SPINDLE_PWM_ON_D8
#define SPINDLE_PWM_ON_D6
//#define SPINDLE_PWM_ON_D9

akkarawat2000, 2023-10-10:
you connect with arduino board direct right ? 

I connect to rampboard . I must change connect to connect arduino board like you ?