fra589/grbl-Mega-5X #344: 5v stepper motor 28BYJ-48 can use in X-Axis in Ramp Board v 1.5 ?

akkarawat2000, 2023-09-22:
Hi 

Now I use 5v stepper motor 28BYJ-48 with Ramp Board V1.5  in  X - Axis Step motor channel 

https://www.youtube.com/watch?v=zuO79Y35Lms

I use  L9110S with 5v stepper motor 28BYJ-48  but now I have problem after i send G code like this 

G1G91X-5.F200  

Result motor not rotate but I touch step motor inside have vibration look like working .

I would like to know I can use with  5v stepper motor 28BYJ-48 with GRBL Mega 5X ? 

config i set like this 

$100=400     Axis 1 (X) steps/unit (mm or deg)
$120 = 50.000   Axis 1 (X) Acceleration, unit/sec^2

I don't understand this how to setup but I try to change $100 many value  but still not rotate but have vibration inside motor

specification for 5v stepper motor 28BYJ-48    = 1/64 

please suggest me how to fixed this problem .

Thank you.

svenhb, 2023-09-22:
I assume you are using 5V with this driver?
I don't think a L9110S is the best choice to drive a stepper motor, better use a real stepper motor driver like A4988.
Whats about your ramps-board? Doesn't already uses A4988 drivers?
Supply the driver with 12V or more, adjust the current and you motor will rotate...

akkarawat2000, 2023-09-22:
> A4988 drivers
Hi , Thank you for reply 

please check my video here 

https://www.youtube.com/watch?v=5Zt7PjqqJ5k

I do correct ? please suggest me. 

and for config in GRBL .  I must change it too ? 

$100=400 Axis 1 (X) steps/unit (mm or deg)
$120 = 50.000 Axis 1 (X) Acceleration, unit/sec^2

and What's value can work with  5v stepper motor 28BYJ-48  ( 1/64 ) 

Thank you very much

svenhb, 2023-09-22:
Sorry I didn't watch the whole video, I'm just wondering why you don't connect the motor directly with the driver, why you put the  L9110S inbetween?
And where are the other cables going to? Other stepper motors connected?
[image]

svenhb, 2023-09-22:
This table means, you need 64 full steps for one turn:
[image]

akkarawat2000, 2023-09-23:
> Sorry I didn't watch the whole video, I'm just wondering why you don't connect the motor directly with the driver, why you put the L9110S inbetween? And where are the other cables going to? Other stepper motors connected? [image]

Why I don't connect 5v Step motor connect to A4988 direct because I see from another video from youtube must convert from unipolar to bipolar first that mean i must make 5 wire now convert to 4 wire but I not have skill like that .

and I'm not sure too if i connect direct with 5 wire with A4988 possile ? 

please check this video 

https://www.youtube.com/watch?v=kCDGEezyzfw

https://www.youtube.com/watch?v=FLJGCZslOcM

Are you sure I can connect 5v step motor to A4988 direct and working with no problem  and I must adjust voltage too ?

-----------------------
And where are the other cables going to? Other stepper motors connected?

It's connect with 12V Step motor working fine but now I just want to test 5V in X-Axis  . That's why i use 5v connect in X-Axis

---------------
What do you want to connect to the motor?
I would start with $100=64 results to 1 turn / mm (if no microstep is activated)
$120 = 100 I would set. You can increase later.

I want just 5v Step motor rotate normally   but I don't understand  how to calculate value for use in $100 and $120 

for my 5V step motor now I just use $100=64  ,  $120 = 100 right ? 

---------------

please suggest me how should i do next step for make 5v Step motor working 

Thank you for help me and reply to me

svenhb, 2023-09-23:
>must convert from unipolar to bipolar 
> if i connect direct with 5 wire with A4988 possile ?

No because the center wires would influence the currents inside the coils. If you have any chance to separate the center wires, you could connect the motor directly to your driver (just don't connect the separated center wires):
[image]
Then you could use microsteps at the driver.

If you don't do this motor modification:
I think you have to disable microsteps at the A8944 to get full signal level to the L9110S.
I would look for an other motor to avoid this level conversion with the additional driver.

>I want just 5v Step motor rotate normally 

It depend on the use, for what is "normal". Do you use it as rotation axis? Then you would assume "unit" as angle-degree and set $100 to 64/360*microstep

After writing this I watched the other videos: exactly what I told :-)

fra589, 2023-09-24:
Hi @akkarawat2000,  
  
The L9110S is not the solution for your need.

> Are you sure I can connect 5v step motor to A4988 direct and working with no problem and I must adjust voltage too ?

After completely making the modification shown here: https://www.youtube.com/watch?v=FLJGCZslOcM
Removing the red wire and cutting the link on the printed circuit, you will be able to connect the motor directly to the A4988 stepper drivers.
  
This is not an issue from grbl-Mega-5X, but only from your hardware choice.
I haven't experience with the stepper motors you want to use, what I am sure of is that the grbl-Mega-5X code is not the cause of your problems with these motors .

@++;
Gauthier.

closed 2023-09-24 by fra589