fra589/grbl-Mega-5X #347: emergency button

ariefadha, 2023-11-09:
sorry for a silly question. Is there any recommendations for wiring emergency stop button?

Is reset pin and power supply cutoff is a good idea?
or hold feed pins?

fra589, 2023-11-10:
Hi @ariefadha,  
  
Yes, you are right! The reset pin (pin A9 by default in grbl-Mega-5X) is the normal way to connect an emergency stop button.
In addition on my own machine, the button also cutoff steppers motors and spindle power.
The Arduino board stay powered, so I can continue to send commands to query it.

@++;
Gauthier

ariefadha, 2023-11-19:
> Hi @ariefadha,
>
> Yes, you are right! The reset pin (pin A9 by default in grbl-Mega-5X) is the normal way to connect an emergency stop button. In addition on my own machine, the button also cutoff [...]

thanks for confirming. there is some cons doing this setup. 1st we need extra long main power wire from back to front where emergency button placed. 2nd reset power can detect when wire is failure. unlike limit switch that can be logic-invert when wire is broken the limit will go high by resistor pullup inside atmega. but I guess this setup is the best that I can have

fra589, 2023-11-19:
Hi @ariefadha,  
  
You can also use a relay to control the main power supply and thus there is no need of extra long main power wire from back to front...

@++;
Gauthier.

ariefadha, 2023-11-20:
> Hi @ariefadha,
> 
> You can also use a relay to control the main power supply and thus there is no need of extra long main power wire from back to front...
> 
> @++; Gauthier.

yeah thats cross in my mind too, so main dc power can keep short. thanks mate 👍

closed 2023-11-20 by ariefadha