fra589/grbl-Mega-5X #348: is frequently change $ value will worn out EEPROM?

ariefadha, 2023-11-18:
for example if my machine has laser and spindle. is changing between "laser mode" will worn out the atmega EEPROM?

svenhb, 2023-11-18:
More than 100000 cycles...
Here is a discussion about: https://arduino.stackexchange.com/questions/226/what-is-the-real-lifetime-of-eeprom#:~:text=ATMEL%20says%20the%20cell%20lifetime,about%20100%2C000%20write%20cycle%2F%20cell.

ariefadha, 2023-11-19:
> More than 100000 cycles... Here is a discussion about: https://arduino.stackexchange.com/questions/226/what-is-the-real-lifetime-of-eeprom#:~:text=ATMEL%20says%20the%20cell%20lifetime,about%20100%2C000%20write%20cycle%2F%20cell.

yeah I know that, ever wonder why shutter in DSLR camera worn out? its because eeprom. for hobbies might need hundred of years to reach that. but for semi manufacturing machine it can reach 3-5 years.

but replacing atmega is easy task for me tho. just have some OCD asking this.

closed 2024-06-30 by fra589