fra589/grbl-Mega-5X #349: Homing cycle doesn't finish x homing

gustavoriverag, 2023-11-22:
I'm having an issue with a cnc lathe, where the homing cycle doesn't finish after x first finds the limit switch. 
The weird thing is it'll do each axis individually ($HZ and $HX) correctly, but the $H command leaves the board frozen until it's reset. 

I have every other limit switch pin (X min, Z min and Y min/max) connected to ground, since i'm using NC switches, and compiled the firmware without the homing cycle for Y, so it shouldn't be getting caught in that. 

Any ideas on what could be causing this?

fra589, 2023-11-22:
Hello @gustavoriverag,  

Can you post your config.h file and the output of $I and $$ ?

@++;
Gauthier.

gustavoriverag, 2023-11-22:
Here's my config.h:

```
/*
  config.h - compile time configuration
  Part of Grbl

  Copyright (c) 2017-2022 Gauthier Briere
  Copyright (c) 2012-2016 Sungeun K. Jeon for Gnea Research LLC
  Copyright (c) 2009-2011 Simen Svale Skogsrud

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

// This file contains compile-time configurations for Grbl's internal system.
// This file is used to define the number of axix and their names,
// define homing sequences or specific needs, i.e.
// performance tuning or adjusting to non-typical machines.

// IMPORTANT: Any changes here requires a full re-compiling of the source code to propagate them.

#ifndef config_h
#define config_h

#include "grbl.h" // For Arduino IDE compatibility.

// Serial baud rate
#define BAUD_RATE 115200

// Define CPU pin map and default settings.
// NOTE: OEMs can avoid the need to maintain/update the defaults.h and cpu_map.h files and use only
// one configuration file by placing their specific defaults and pin map at the bottom of this file.
// If doing so, simply comment out these two defines and see instructions below.
#define DEFAULTS_GENERIC
#define CPU_MAP_2560_RAMPS_BOARD

//----------------------------------------------------------------------
// Axis definitions :
//----------------------------------------------------------------------
// IMPORTANT: When changing the axis definitions (axis numbers N_AXIS, 
// linears axis number N_AXIS_LINEAR or axes names AXIS_*_NAME, 
// don't forget to issue the reset factory defaults Grbl command: $RST=*
// if you forget the $RST=* command after change, Grbl may have 
// unpredictable behavior!
//----------------------------------------------------------------------

#define N_AXIS 3        // Number of axes (3 to 6)
#define N_AXIS_LINEAR 2 // Number of linears axis, must be <= N_AXIS

// Axis indexing and names
#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V, W, D, E & H.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'
#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
[... 751 lines cut ...]
$104 = 8.889   
$110 = 750.000    (X-axis maximum rate, mm/min)
$111 = 750.000    (Y-axis maximum rate, mm/min)
$112 = 750.000    (Z-axis maximum rate, mm/min)
$113 = 1440.000   
$114 = 1440.000   
$120 = 5000.000    (X-axis acceleration, mm/sec^2)
$121 = 50.000    (Y-axis acceleration, mm/sec^2)
$122 = 1000.000    (Z-axis acceleration, mm/sec^2)
$123 = 50.000   
$124 = 50.000   
$130 = 400.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
$133 = 360.000   
$134 = 180.000 
```

And my $I:
```
$I
[VER:1.2h.20220109:]
[AXS:5:XYZAB]
[OPT:VNMGH,35,255,88]
```

fra589, 2023-11-22:
Hi @gustavoriverag,  

The response from the $IP command does not match your config.h file.
`[AXS:5:XYZAB]` => 5 axes with names X, Y, Z, A and B. This come from default config.h.

You probably missed the grbl-Mega-5X compilation and upload procedure:

From the wiki : https://github.com/fra589/grbl-Mega-5X/wiki/Compiling-grbl-Mega-5X
_Compiling GRBL or Advanced Users: Most users are just fine with Grbl's default build, but you can customize Grbl by editing the config.h file. It is **extremely important** to edit the files **inside** the Arduino _`library`_ folder not the folder you imported the grbl from._

@++;
Gauthier.

gustavoriverag, 2023-11-22:
That's probably ut lol, will try, thank you very much 😭

fra589, 2023-11-23:
Don't forget to close issue if it work for you...

gustavoriverag, 2023-11-23:
Just tested it, and confirm that was the issue. Thanks :)

closed 2023-11-23 by gustavoriverag