fra589/grbl-Mega-5X #350: I need some help

srmarkovic, 2023-12-12:
I made custom board for Atmega2560 grbl controller for my hobby machine. I used pinouts shown in the bottom picture. It worked fine with grbl/Mega. I need 4th axe so I tried your 5X grbl. It works fine, does homing, shows when limit switches (inductive with isolated 24V psu, shielded cable optocouplers and smit trigger circuits working beautiful) are triggered but doesn't halt doesn't invoke Alarm.  I tried with  "#define ENABLE_RAMPS_HW_LIMITS " and it cant send pulses it stucks or does something strange. 
Please help me. Where did I go wrong. Thank you. Here is my cpu_map.h

/*
I made custom board for grbl. 
I used next pins :
Limit Z             12
Limit Y             11
Limit X             10
Mist                 9
Flood                8 
Spindle PWM          7    
Spindle enabled      6
Spindle direction    5
Pulse X             24   
Pulse Y             25
Pulse Z             26
Pulse A             27
Pulse B             28
Pulse C             29
Direction X         30
Direction Y         31
Direction Z         32
Direction A         33
Direction B         34
Direction C         35
Limit C             51
Limit B             52
Limit A             53
Probe              A15
Safety door        A11
Start              A10
Hold                A9
Reset               A8                               
*/
/*
  cpu_map.h - CPU and pin mapping configuration file
  Part of Grbl

  Copyright (c) 2017-2022 Gauthier Briere
  Copyright (c) 2012-2016 Sungeun K. Jeon for Gnea Research LLC

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

/* The cpu_map.h files serve as a central pin mapping selection file for different
   processor types or alternative pin layouts. This version of Grbl supports only the
   Arduino Mega2560. */

#ifndef cpu_map_h
#define cpu_map_h

#ifdef CPU_MAP_2560_RAMPS_BOARD // (Arduino Mega 2560) with Ramps 1.4 Board
  #include "nuts_bolts.h"

  // Serial port interrupt vectors
  #define SERIAL_RX USART0_RX_vect
  #define SERIAL_UDRE USART0_UDRE_vect

[... 550 lines cut ...]
      #endif
      #define OUTPUT_PWM_OFF_VALUE     0
      #define OUTPUT_PWM_RANGE         (OUTPUT_PWM_MAX_VALUE-OUTPUT_PWM_MIN_VALUE)
      //Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
      #define OUTPUT_TCCRA_REGISTER    TCCR4A
      #define OUTPUT_TCCRB_REGISTER    TCCR4B
      #define OUTPUT_OCR_REGISTER      OCR4A
      #define OUTPUT_COMB_BIT          COM4A1
      // 1/8 Prescaler, 8-bit Fast PWM mode
      #define OUTPUT_TCCRA_INIT_MASK (1<<WGM41)
      #define OUTPUT_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
      #define OUTPUT_OCRA_REGISTER   ICR4 // 8-bit Fast PWM mode requires top reset value stored here.
      #define OUTPUT_OCRA_TOP_VALUE  0xFF // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.
      // Define spindle output pins.
      #define OUTPUT_PWM_DDR   DDRH
      #define OUTPUT_PWM_PORT  PORTH
      #define OUTPUT_PWM_BIT   3 // MEGA2560 Digital Pin 6
    #else
      #error "OUTPUT_PWM_ON_D9 or OUTPUT_PWM_ON_D8 or OUTPUT_PWM_ON_D6 must be defined in config.h with the USE_OUTPUT_PWM option!"
    #endif
  #endif // USE_OUTPUT_PWM
#endif // CPU_MAP_2560_RAMPS_BOARD
#endif 

[image]

RaphaelDives, 2023-12-12:
Hello!
Just a quick (stupid) question:
Did you think of enabling hard limits? ($21 - Hard limits, boolean).
I think they are dsiabled by default...

Cheers
Raphael

srmarkovic, 2023-12-12:
> Hello! Just a quick (stupid) question: Did you think of enabling hard limits? ($21 - Hard limits, boolean). I think they are dsiabled by default...
> 
> Cheers Raphael
They are enabled. 

When I don't use #define ENABLE_RAMPS_HW_LIMITS  everything works and it is detected when limit switch is active (show 
RUN (x), RUN(y), RUN(Z)  according to axe limit activated  in BCNC  or when  it is not in run state and I ask $$ )  but there is no Alarm and machine continues to work. When I use #define ENABLE_RAMPS_HW_LIMITS there is alarm and machine stop when limit is activated but it loses steps jogging or stop or uncontrolled continue to go in one direction and do all kind of weird things and it is as it has no power.

closed 2024-06-30 by fra589