fra589/grbl-Mega-5X #354: PWM on D8

Zjokkeh, 2023-12-27:
Hi,

After using grbl mega for several months now and beeing very happy with it, I thought it was time for a laser module. So I bought a NEJE   N30820 module. The manual tells me that the ttl is between 2.5V ~ 12V DC. This is what D8 should deliver. I have tried everything but I can only get 12V max and 8.3V min. And it is inverted. M3S1000 gives me 8.3v and M3S0 gives me 12V. Any thoughts on this? 

Thanks in advance and happy holidays!

Some TTL info:
[image]

fra589, 2023-12-28:
Hi @Zjokkeh,  
  
Can you please, tell me more info about your problem?
- Send the output of Grbl commands `$$` and `$I`,
- Do you made any change in files config.h, cpu_map.h, default.h?
- Tell me your gcode command which give you max 12v on D8 and min 8.3v
  
I need a maximum of informations to studdy your problem...

@++;
Gauthier.

fra589, 2024-02-25:
No reply to additional infos... => Close issue

closed 2024-02-25 by fra589