fra589/grbl-Mega-5X #355: Duplicate Axis Issues

ShadowGoat128134, 2024-01-01:
Hi! I'm having some issues with duplicate axes that might be bugs. My machine is set up as a 5 axis machine with 3 of them being configured as part of the Z Axis. Axis 4 is unwired just because the way the board was set up it had an inaccessible stepper driver configured on the stepper 4 pins and I didn't want to deal with changing the pin definitions, though I can if that ends up being part of the problem. The issue I'm having is that axis 5 is completely unresponsive to inverting the axis or homing direction, and ignores hard limits. Attempting to flip the travel/homing direction for the z axis does not affect axis 4 or 5, and attempting to home the z axis results in axis 3 stopping upon triggering the limit switch but axis 5 does not. I have attempted to change the endstop pins in CPU_MAP.h with no success, and for some reason changing MAX_LIMIT_BIT_3 and MAX_LIMIT_BIT_5 to be the same, which I saw in GRBL documentation is what I should change, only results in the y axis stepper becoming unresponsive for some reason and does not affect the issue with hard limits. I would like to also configure single pin homing for the Y and Z axis due to how the board is set up, but I have been unable to get this working as well. Either I'm changing something wrong or there's something weird about my setup. Hard limits are turned on. Maybe this is the wrong place for this but I would appreciate assistance troubleshooting. I can provide $$ outputs and code screenshots as needed but I'm not directly at the machine as of writing this.

fra589, 2024-01-02:
Hi @ShadowGoat128134,  
  
1st question: Do you have issued the Grbl's command: `$RST=*` (reset factory parameters) when you have changed yours axes parameters ?
If no, grbl-Mega-5X may have very unpredictable behaviors when the stored flash parameters does not agree with the axes definition!

Then, to studdy your problem, I will need the output of Grbl commands `$I` and `$$`, and copy of all your modifieds configuration files.

@++;
Gauthier.

ShadowGoat128134, 2024-01-02:
Doing $rst=* fixed the weird axis behavior but did not resolve the issue with single pin homing and the duplicate z axis not responding to the main z axis endstop. Here's my $$:
`$0 = 10    (Step pulse time, microseconds)
$1 = 254    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 6    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.020    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 250.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 12000    (Maximum spindle speed, RPM)
$31 = 550    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 80.000    (X-axis travel resolution, step/mm)
$101 = 80.000    (Y-axis travel resolution, step/mm)
$102 = 400.000    (Z-axis travel resolution, step/mm)
$103 = 400.000   
$104 = 400.000   
$110 = 187.500    (X-axis maximum rate, mm/min)
$111 = 187.500    (Y-axis maximum rate, mm/min)
$112 = 187.500    (Z-axis maximum rate, mm/min)
$113 = 187.500   
$114 = 187.500   
$120 = 50.000    (X-axis acceleration, mm/sec^2)
$121 = 50.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$123 = 50.000   
$124 = 50.000   
$130 = 400.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
$133 = 200.000   
$134 = 200.000  `

Here's $I:
`[VER:1.2h.20220109:]
[AXS:5:XYZZZ]
[OPT:VNMGH,35,255,56]`

The files I've tweaked are config.h and cpu_map.h. The main sections I've tweaked are defining new axes and homing cycles in config.h and trying to rebind pins for single pin homing under the "homing/hard limit switch input pin" section of cpu_map.h. The behavior I am attempting to get is that axis 5 uses the Z_Max pin for homing since it's part of the z axis, and Axis 2 also uses the Z_max pin for homing since the Y_min endstop is wired to the same pin. I have Z and Y running in separate homing cycles so they shouldn't interfere. Thank you for any help you can provide!
[troubleshooting.zip](https://github.com/fra589/grbl-Mega-5X/files/13811676/troubleshooting.zip)

fra589, 2024-01-12:
Hi @ShadowGoat128134,  
  
Sorry for the late reply, I was overbooked las 2 weeks...

If you plan to using the same limit switch for your 3 Z axes, you need to define this in the **cpu_map.h** like this:
(assuming the only one whitch is the standard Z one (min & max))
(ports & bits are numbered from 0 to N_AXIS - 1 in cpu_map.h)
```
#define MIN_LIMIT_PORT_2 D
(...)
#define MIN_LIMIT_PORT_3 D
(...)
#define MIN_LIMIT_PORT_4 D
(...)
#define MIN_LIMIT_BIT_2 3
(...)
#define MIN_LIMIT_BIT_3 3
(...)
#define MIN_LIMIT_BIT_4 3
(...)
#define MAX_LIMIT_PORT_2 D
(...)
#define MAX_LIMIT_PORT_3 D
(...)
#define MAX_LIMIT_PORT_4 D
(...)
#define MAX_LIMIT_BIT_2 2
(...)
#define MAX_LIMIT_BIT_3 2
(...)
#define MAX_LIMIT_BIT_4 2
```

So, the 3 cloned axes will stop in the same time than the only one switch will be triggered.

@++;
Gauthier.

ShadowGoat128134, 2024-01-12:
Looking at my cpu_map.h I can see that I seem to have it configured the same, with the exception of the min limit pins because I'm not using them. Do I need to set both in order for it to function? My max limit pin config is set identically to what was shared above.

fra589, 2024-01-13:
Hello @ShadowGoat128134,  

As is from the original version of gnea/grbl, limit check algorithm doesn't make difference between the max and min limit switch.
In fact, the limit code check both min and max switch status to verify the limit.
This why you should have both min and max config correctly cloned .

@++;
Gauthier.

ShadowGoat128134, 2024-01-13:
Alright, that makes sense and explains what I was doing wrong. Thanks!

closed 2024-06-30 by fra589