fra589/grbl-Mega-5X #357: PROBLEME de connexion avec arduino mega

franck70150, 2024-01-13:
bonjour j'ai un problème de connexion avec mon arduino mega 2560 origine
elle est reconnu sur lasergrbl sur cn5x++ mais pas sur lightburn 

sur cn5x++
au démarrage elle m'affiche ce si   :      2024-01-13 08:00:16 : Info    : grblCom: Starting grblComSerial thread on COM11.
2024-01-13 08:00:16 : Info    : grblComSerial.run(): Running "grblComSerial" from thread #0x1b78.
2024-01-13 08:00:16 : Info    : grblComSerial.__openComPort(): comPort COM11 open.
2024-01-13 08:00:18 : Info    : cn5X++ : Grbl initialisé.
2024-01-13 08:00:18 : Warning : grblComSerial.__mainLoop(): utf-8 decode error, buff=b'[VERLI:\xb92220917:]\r\n'
2024-01-13 08:00:18 : Warning : grblComSerial.__mainLoop(): utf-8 decode error, buff=b'[AXS:4:XYZA]\xa4[OPT:95\x1di!\xb135,255,18]\r\n'

peut tu me dire ou j'ai un erreur et pourquoi sur lightburn je n'arrive pas a me connecter

merci d'avance

fra589, 2024-01-13:
Bonjour @franck70150 

Les "Warning" que te montrent la trace de cn5X++ me font penser à des erreurs de communication série...
Dans cn5X++, ce type d'erreurs de communication ne sont pas considérées bloquantes et le programme arrive quand même à travailler malgré ça.

Je ne connais pas précisément lightburn, il est peut être plus sensible à ce type d'erreurs...

Dans cn5X++, que renvoi la commande `$I` lorsque tu l'envoi manuellement ? Cela génère-t-il les mêmes "Warning" ?

@++;
Gauthier.

franck70150, 2024-01-19:
merci de votre réponse 
si je tape $I non je n'ai pas de warning d'après la toile je ne suis pas le seul a avoir des problèmes de connexion avec lightburn je pense avoir résolu car ca fait deux soir que je connecte sens problème .
merci de votre réponse je trouvais bizarre tous c'est warning .
j'ai une nouvelle question je monte un laser co2 j'utilise (LASER_PWM_ON_D9) dans la config  faut il laisser $=30 a 255 ou peut on le maitre a 1000  car j'ai bien les dégradé de puissance  mais quand je lui dis 100% j'ai que 4.38volt de sortie  je sais pas si vous me comprenez et y a t il une ligne spécifique de délais du démarrage du laser et inversement quand le laser s'arrête et repars j'ai des brules plus forte merci je commence a comprendre grbl mais il y a beaucoup de ligne qui ressemble a du chinois encore 
merci bon week end et bravo pour le boulot salutation
@++

fra589, 2024-01-21:
Bonjour @franck70150,  
  
Quand tu utilises `LASER_PWM_ON_D9` en ayant également défini `SEPARATE_SPINDLE_LASER_PIN`, c'est `$33` qui définit le niveau de puissance maximum du laser.

Dans ce mode, $30, c'est pour définir la vitesse maxi de la broche et non du laser.

Ce qui veut dire que si (après avoir configuré $32 = 1 pour activer le mode laser) tu utilises `M3 S255` cela activera le laser à 100% de la capacité électrique de la pin D9 (`S127` activera 50% de la sortie D9).

Donc, avec `$33=255`, si tu utilises le Gcode `M3 S255`, tu devrais obtenir la tension max sur la pin D9 soit 100% de la sortie. Si tu n'as que 4,38 volts, c'est que : soit ton alimentation ne délivre pas plus que cette tension, soit que ton laser consomme trop de puissance et que la carte Arduino n'est pas en mesure de fournir la puissance nécessaire.

par défaut, `$33` vaut 255 car le timer qui contrôle la sortie PWM D9 au sein du microcontroleur de la carte Arduino est un timer 8 bits. Et avec 8 bits, il n'y à que 255 niveaux de sortie différents possible. si tu utilises `$33=1000` (c'est possible mais), Grbl calulera la valeur entière la plus proche avec un formule qui ressemble à `Puissance = entier(Valeur_S / 1000 x 255)`.

@++;
Gauthier.

franck70150, 2024-01-25:
bonjour Gautier 
j'ai bien vue le SEPARATE_SPINDLE_LASER_PIN je crois y ai que ca marchais qu'avec une machine double  .
j'ai bien configuré le SEPARATE j'ai bien $33 et $34 
dans config  il faut bien configurer pwm broche , pwm laser et  USE_OUTPUT_PWM et qu'il sois tous diffèrent 
j'ai encore un problème de gravure il brule les extérieurs en m3  et ne marque pas les extérieurs en m4 .
j'ai regarder le programme ils sont bon . est ce que la broche use output est bien le déclencheur du laser ?et y a t'il une ligne pour l'inverser .j'ai juste trouvé #define INVERT_SPINDLE_ENABLE_PIN qui me donne une erreur quand je l'utilise .

je ne comprends pas  les connectiques de l'alimentation du laser co2  il me faut un IN pwm et un contact L  pour le déclanchement  avec une masse ca marche mais ca brule j'ai cherché sur le net mais ils ne disent jamais la même chose .

je vous répons tardivement car je cherche de mon coté . merci pour votre aide 

salutation

fra589, 2024-01-26:
Bonjour @franck70150,  
  
` USE_OUTPUT_PWM` est une option pour ajouter une sortie PWM auxiliaire supplémentaire gérée par les commandes GCode `M67` et M68 (http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m67).

`SEPARATE_SPINDLE_LASER_PIN`, c'est pour avoir des sorties différentes entre la broche et le laser pour les machines doubles (broche + laser). Dans ce cas, la broche et le laser sont tous les deux gérés par les commandes GCode `M3`, `M4` et `M5`. Le choix de la broche de sortie se fait selon l'état du paramètre Laser-mode enable ($32). 

Effectivement, toutes les sorties PWM ayant des fonction différentes (broche, laser et sortie auxiliaire) doivent être sur des pins différentes.

En mode fraisage (`$32=0`), la différence entre `M3` et `M4`, c'est le sens de rotation de la broche. Au niveau matériel, ce sens est géré par la sortie à l'état 0 ou 1 de la pin "Spindle direction" (D5 par défaut).

En mode laser (`$32=1`), le mode constant (`M3`) va appliquer strictement les consignes de puissance du laser contenu dans les instructions (Gcodes Sxxx). Le mode dynamique (`M4`) va adapter la puissance du laser en fonction de la vitesse pour avoir le même résultat de gravure ou de découpe lors des phases d'accélération/décélération et de vitesse constante. Cela évite les brûlures en début et fin de ligne.

Au sujet de la connectique d'alimentation du laser co2, le IN PWM, c'est la broche que tu as définie (D9 si `LASER_PWM_ON_D9` dans config.h) et le contact que tu appelles "L" correspond à la broche "Spindle enable" (définie sur D4 par défaut) la même pin sert à la mise en route (déclenchement) de la broche ou du laser.

@++;
Gauthier.

franck70150, 2024-01-26:
Merci pour votre aide j' essaie ce soir je vous redis   bonne journée

closed 2024-06-30 by fra589