fra589/grbl-Mega-5X #361: Laser pin pinout (sorry to ask again)

GeoffreyOlivier, 2024-02-15:
I don't know what to do exactly, for set the pinout for my laser 12V.

I see lot of part of solution, but I don't know what to do exactly, and why. Can you help me please ?

Is for what, this ?
```
line 341 cpu_map
   #define SPINDLE_OCR_REGISTER OCR4B
   #define SPINDLE_COMB_BIT COM4B1
   #define SPINDLE_PWM_BIT 4

```
I have 3 SPINDLE_OCR_REGISTER in my code. 

I need ti change that to ? But I have multiple to, I don't know wich change ?
```
    // Define spindle output pins.
    #define SPINDLE_PWM_DDR   DDRH
    #define SPINDLE_PWM_PORT  PORTH
    #define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 8

```

fra589, 2024-02-16:
Hello @GeoffreyOlivier 

Perhaps I don't have understand precisely your problem?
Knowing which laser module you want to control and describing the behavior which doesn't work as you want would be interesting to be able to help you more effectively...

When grbl-Mega-5X is used with a 12V powered RAMPS board, the Spindle PWM output (on pin D8 by default) normally provides power between 0 and 12V.

To do this, there is nothing to modify in the default configurations.

@++;
Gauthier.

GeoffreyOlivier, 2024-02-19:
Thank you for your answer.

The laser doesn't light up, here is a diagram of my wiring.
You tell me to wiring in the green cercle ?
I wiring postive and neg on the cercle red, and the PWM on D11 or D6.
[image]

fra589, 2024-02-20:
I suspect a misunderstanding on your part regarding the connection of the laser...

Usually, a laser is powered by 12V by connecting it directly to the 12V power supply,
Then, its power is managed by a 0-5V PWM output.

In this case, you must connect the 12V power supply of your laser directly to the + & - power supply.
Then, in config.h, you set the PWM output of the Spindle to D6:
```
//------------------------------------------------ ----------------------------
// Spindle, laser and other PWM output
//------------------------------------------------ ----------------------------
// Chose the spindle pin output:
// SPINDLE_PWM_ON_D8 => 0-12v 16 bits PWM on RAMPS D8 (default)
// SPINDLE_PWM_ON_D9 => 0-12v 8 bit PWM on RAMPS D9
// SPINDLE_PWM_ON_D6 => 0-5v 8bits PWM on RAMPS Servo 2 signal (Mega 2560 D6)
// Uncomment the line which corresponds to your hardware
//#define SPINDLE_PWM_ON_D8
#define SPINDLE_PWM_ON_D6
//#define SPINDLE_PWM_ON_D9
```  

And you connect the PWM pin of your laser to pin D6 with ground as in the image below:
[image]

Can you tell us the reference of your laser?
A link to its documentation would allow us to check if what I just wrote is correct.

@++;
Gauthier.

fra589, 2024-02-20:
OH! Reading your laser user manual, I saw that it need 0-1V PWM signal!

So, you must leave the spindle PWM signal on D8:
```
//------------------------------------------------ ----------------------------
// Spindle, laser and other PWM output
//------------------------------------------------ ----------------------------
// Chose the spindle pin output:
// SPINDLE_PWM_ON_D8 => 0-12v 16 bits PWM on RAMPS D8 (default)
// SPINDLE_PWM_ON_D9 => 0-12v 8 bit PWM on RAMPS D9
// SPINDLE_PWM_ON_D6 => 0-5v 8bits PWM on RAMPS Servo 2 signal (Mega 2560 D6)
// Uncomment the line which corresponds to your hardware
#define SPINDLE_PWM_ON_D8
//#define SPINDLE_PWM_ON_D6
//#define SPINDLE_PWM_ON_D9
```

Then connect it like this:  
[image]

@++;
Gauthier.

GeoffreyOlivier, 2024-02-20:
I reinstalled your GRBL, made the new wiring, and changed $100 and $101. When I turn on the CNC, the laser fan spins, which seems pretty logical to me.

But in LightBurn, it's impossible to make the laser light emit. Even with the commands M3, M4, M4 S200, etc.

Thanks for your help

fra589, 2024-02-20:
If you have reinstalled a fresh grbl-Mega-5X without any changes in the config.h, default.h and cpu_map.h, you should begin by issuing the grbl `$RST=*` command to also reset the flash memory to the default configuration.
Then, redoing your $$ config.
- $100 and $101 are just for X and Y axis travel resolution,
- You need to set `$32=1` to enable the laser mode,
By default, the maximum value for the laser S value is 255, this give the laser pin 100% of PWM signal.
the default value for the min laser value is 1.

So, if you issue `M3S255` or `M4S255`, you should have 12V output on the pin D8,
if you issue `M3S127`, you should have about 6V.
You can adjust those values with $33 Maximum laser value and $34 Minimum laser value.

But for security reason, the laser output is only active when moving!
So, if you want to test your laser, you need to do it with one or more axis move.

The difference between M3 and M4 is M3 power the laser at the desired power immediately when moving, M4 will power the laser gradually, according the axis acceleration.

@++;
Gauthier.

GeoffreyOlivier, 2024-02-22:
Thanks for your help, but it isn't working.

I tried using LaserGRBL on another computer, but the laser doesn't light up, even on "show focus". I don't understand why it's so difficult. I've tried two different GRBLs and one Marlin, and I encounter different problems with each.

fra589, 2024-02-23:
Hello @GeoffreyOlivier,  

> Thanks for your help, but it isn't working.

"it isn't working" is not an understandable computer problem!?!?!?
More precises info are needed to understand your problem :-(
- config.h, cpu_map.h & default.h: not modified or some adjustment?
  - If you have made modification in one of those file, post your modification
- $$ output,
- $I output,
- expected behavior,
  - command issued which doesn't give the expected behavior,
  - obtained behavior,
  - error message or no error given by Grbl,
- Do you issued the reset flash memory Grbl's command (`$RST=*`) when you have changed the Grbl

... And any information that could direct us towards the solution.

> I don't understand why it's so difficult.

Maybe because it is a complex technology? Some people have studied for a long time before mastering these technologies and making it their profession. This requires a minimum of knowledge and you have to take the time to learn it.

> I've tried two different GRBLs and one Marlin

- Do you issued the reset flash memory Grbl's command (`$RST=*`) when you have changed from one Grbl to another and from Marlin to Grbl? And when you have changed the compil time Grbl's configuration (modify any source file and reflash Grbl)? Without the `$RST=*`, Grbl may have unpredictable behavior...

> Additionally, I executed the command M3S255 and measured the voltage with a multimeter, but there's nothing on D9.

Testing individual function behavior: Good!
But as I wrote, the default output of spindle / laser is D8 and not D9.

@++;
Gauthier.

GeoffreyOlivier, 2024-02-23:
> config.h, cpu_map.h & default.h: not modified or some adjustment?

No modification

> $$ output,

$$
$0=10
$1=254
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=25.000
$25=250.000
$26=250
$27=5.000
$30=12000
$31=550
$32=1
$100=160.000
$101=160.000
$102=400.000
$103=8.889
$104=8.889
$110=750.000
$111=750.000
$112=750.000
$113=1440.000
$114=1440.000
$120=50.000
$121=50.000
$122=50.000
$123=50.000
$124=50.000
$130=400.000
$131=200.000
$132=200.000
$133=360.000
$134=180.000

> $I

[VER:1.2h.20220109:]
[AXS:5:XYZAB]
[OPT:VNMGH,35,255,18]
Target buffer size found
ok

> expected behavior

Start laser cut, basic cercle :
[image]
The cnc move, but the laser don't light up.

 M3S255 or M4S255 -> console response is ok, but the laser don't light up

> Do you issued the reset flash memory Grbl's command ($RST=*) when you have changed the Grbl

yes, multiple time

>Testing individual function behavior: Good!
>But as I wrote, the default output of spindle / laser is D8 and not D9.

Is was a mistake, It's wiring on D8, but I tested on D8, D9, D10.

thanks for your help

fra589, 2024-02-24:
Hi @GeoffreyOlivier,  
  
Sorry, I forgot a little detail in my previous answers...
When you use unmodified config.h, cpu_map.h & default.h, the defaults max & min values for M3/M4S_XXX_ are respectively `$30=12000` & `$31=550`. This made good values for the spindle, but not for the laser.

So, the Grbl command I tell you to test (`M3S255`) is not good! This give an output on D8 of 0.255V :-( Sorry
(12V / 12000 x 255 = 0.255V)

You must adjust `$30=255` and `$31=1` for it to work.

And also, for fire security reasons, the laser output will be only active during axes moves. So, to test the output, you should do 2 commands:
`M3S255`    
`G1X<value to move>`

Then, test the output during the move.

@++;
Gauthier.

GeoffreyOlivier, 2024-02-24:
Hi, 

I made the change but same problem. There my start of Gcode :
; LightBurn 1.5.00
; GRBL device profile, absolute coords
; Bounds: X11 Y8 to X54 Y60

;USER START SCRIPT
M3S255
;USER START SCRIPT

G00 G17 G40 G21 G54
G90
M4
; Cut @ 1000 mm/min, 80% power
M9
G0 X21.351Y11.764
; Layer VirtArray Pass 1 of 2
G1 X20.479Y12.44S9600F1000

fra589, 2024-02-24:
Ah, OK...

In your GCode start, the `M3S255` line you added do nothing, but it's not a (the) problem :-)

=> 3 lines later, LightBurn use the M4 command to power on the laser with the power modulate by the function of the axes acceleration. This replace the M3 mode by the M4 mode which is the normal mode for Grbl laser.

=> Then, 3 line lower, LightBurn have set the laser power in the G1 move to `S9600`. This replace the `S255` you putted n the start script.

In fact, as I can see, the max power of laser was correctly set at 12000 in the LightBurn device setting => 12000 at 80%power = 9600.

You need to have the same value in the Grbl $30 parameter than in the "S-Value Max" of LightBurn device setting. 
For the rest, I advise you to use 1024 as value for both Grbl $30 and LightBurn S-Value Max. This value is the optimal setting considering the 16bits Grbl register used for PWM generation.

With all those information, I have new questions for you:
- Have you verified the power output on D8 with a multimeter when the laser is moving by the GCode you just posted?
- The LASER TREE LT-80W-AA-PRO exist with 2 optionals powering (not the PWM, but the general power supply): 12V or 24V
  - Is your laser is a 12V powered or a 24V one?
- The laser can be supplied with or without driver adapter (P-DA-01)
  - Do you have connected the laser directly to the Grbl board, or do you have connected Grbl to the driver adapter P-DA-01, then the driver adapter to the laser?

@++;
Gauthier.

GeoffreyOlivier, 2024-02-24:
There my $$

$$
$0=10
$1=254
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=25.000
$25=250.000
$26=250
$27=5.000
$30=1024
$31=1
$32=1
$100=160.000
$101=160.000
$102=400.000
$103=8.889
$104=8.889
$110=750.000
$111=750.000
$112=750.000
$113=1440.000
$114=1440.000
$120=50.000
$121=50.000
$122=50.000
$123=50.000
$124=50.000
$130=400.000
$131=200.000
$132=200.000
$133=360.000
$134=180.000

Yes, I attempted to use the P-DA-01 with the same settings, but encountered the same issues.

Additionally, there's zero voltage on D8 :/

I'm considering switching to an SKR Pro board with Marlin firmware from V1e.com. I'm starting to wonder if the issue might be with the board itself, which seems odd.

fra589, 2024-02-25:
Hello @GeoffreyOlivier,  

Hmmm, I suspect a hardware problem...

First, we can make a last test with multimeter:
Switch off the laser mode: `$32=0`
Then power on the spindle (without the laser security because `$32=0`): `M3S1024`
Check the voltage between D8 and GND with the multimeter.
If it still has 0 V, your hardware is probably broken.

But the default is perhaps limited to the D8 output. So, we can try to switch the output to D9 which is a second 12V PWM output available for grbl-Mega-5X.

To change the output to D9:
- Edit the config.h file and change the lines 123 to 125 like this:
```
//#define SPINDLE_PWM_ON_D8
//#define SPINDLE_PWM_ON_D6
#define SPINDLE_PWM_ON_D9
```

- Recompile and reflash grbl-Mega-5X to your board:
cf. https://github.com/fra589/grbl-Mega-5X/wiki/Compiling-grbl-Mega-5X
- Reset the $$ flash memory config with `$RST=*`
- Reset your own $$ settings, but use `$30=255` (as the D9 PWM output use a 8bit register and not a 16 bits)
- Leave `$32=0`
- Issue spindle on command: `M3S255`
- Check the voltage between D9 and GND with the multimeter.

@++;
Gauthier.

GeoffreyOlivier, 2024-03-02:
I switch to SKR pro 1.2 on Marlin, thanks for you help !!!

closed 2024-03-02 by GeoffreyOlivier