fra589/grbl-Mega-5X #362: No delay after command M3 to keep time "pen down"

Srallie, 2024-02-19:
Hello,

is it possible to create a delay of 300ms standard after command M3 SXX so that the plotter pen has time to go down? By pen up there is no delay needed.  I wanna change it in the firmware, i dont want to change the gcode every time. Do you know how? in gcode.c?

greetz

fra589, 2024-02-19:
Hi @Srallie,  
  
It's probably possible, but in my opinion it's not a good solution. Changing this in Grbl will lose compatibility with standard GCode operation.
The normal solution to do what you need is to use a G4 P0.3 (Dwell GCode: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g4) after the M3 command.
This should be able to be added automatically by the post processor of the software that generates the GCode.

If you still want to add this tempo in Grbl-Mega-5X, you may be able to do it in spindle_control.c, in the spindle_sync() function.

@++;
Gauthier.

Srallie, 2024-02-19:
Hello,

Thank you for your answer. I want to use lightburn for my laser project, and now also for my vinyl cutter,then i can use the same program. But lightburn didnt have the option to tweak the gcode. Thats the reason that i will implement it in the firmware.

When i still want to change spindle_control.c, but i dont know how. I am not be able to figure that out. Small changes i can,but adding lines not.

Greetz lars

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

fra589, 2024-02-19:
Hello @Srallie,  
  
Try to change the spindle_sync() function like this:  
_(from line 306 in the spindle_control.c file)_
```
// G-code parser entry-point for setting spindle state. Forces a planner buffer sync and bails 
// if an abort or check-mode is active.
void spindle_sync(uint8_t state, float rpm)
{
  if (sys.state == STATE_CHECK_MODE) { return; }
  protocol_buffer_synchronize(); // Empty planner buffer to ensure spindle is set when programmed.
  spindle_set_state(state,rpm);
  if (gc_state.modal.spindle != SPINDLE_DISABLE) {
    delay_ms(300); // Force delay of 300 ms.
  }
}
```  
  
I'm not be sure if this will work since don't have enough time and resources to make the needed tests.
Tell me if it's help you :-)

@++;
Gauthier.

Srallie, 2024-02-19:
Hello,

First thanks a lot for your help.
I have tried this code with 1000ms to see what happens. Now the knife goes down, cut a letter,then wait for 1s , then knife up, goes further to the next letter and knife goes directly down without delay, cut the letter,wait 1s and goes further.....

The delay is on the wrong place?
Near by the right place i hope.

Greetz Lars

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

fra589, 2024-02-20:
HI @Srallie,  
  
Oops !... Missed! :-)  
  
By changing the direction of the test, it should probably work ?...  
```
   if (gc_state.modal.spindle == SPINDLE_DISABLE) {
     delay_ms(300); // Force delay of 300 ms.
   }
```  
  
@++;
Gauthier.

Srallie, 2024-02-20:
Hello Gauthier,

Now there is no delay after going up or down. Only after going down it need a pause. (Laser mode is selected)

greetz lars

fra589, 2024-02-20:
Hummmm...

I need to check more code...
Sorry, I don't have lot of time this morning,... I will see it this afternoon...

Srallie, 2024-02-20:
Thank you👍i will check it today after 11pm.

Greetz Lars

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

Srallie, 2024-02-20:
> I'm sorry, I was too quick to respond yesterday and this morning. I hadn't thought enough :-)
> This should work now:

Hello,

He stops 300ms everytime when the knife is back up but not when the knife is down. Maybe it helps when the pwm is set from ** to zero and then pause 300ms?

greetz lars

  if (state == SPINDLE_ENABLE_CW) {

(the delay work only for M3, not for M4 or M5)

@++;

fra589, 2024-02-21:
Hello Lars,
  
So, I didn't understand at what point in GCode commands you want there to be a pause.
I don't know your plotter and what commands control the pen.
In the last code I posted, pause is executed after every M3 command.
Can you explain what are the pen orders?
Can you send an example of the GCode you use to drive the pen indicating where you want to insert the pauses?

@++;
Gauthier

svenhb, 2024-02-21:
> I want to use lightburn for my laser project, and now also for my vinyl cutter

Do you know, that a RC-servo needs a different PWM than a laser?
Do you know, that a drag knife needs a different tool path than a laser?
I don't think lightburn can create a tool path for a drag knife...

Srallie, 2024-02-21:
Hello,

In the nc code:

; LightBurn 1.4.01
; GRBL device profile, absolute coords
; Bounds: X264.71 Y71.16 to X401.29 Y80.84
G00 G17 G40 G21 G54
G90
M4
; Cut @ 6000 mm/min, 51% power
M9
G0 X265.037Y77.862
M3
; Layer 100
G1 X264.987S51F6000    ----------> knife goes down
.....
.....
.....
G0 X270.537Y75.037 -----------> knife goes up and moves to position
G1 X270.438Y74.737-----------> knife goes down and moves

I see that after the M3 – G1 X264.987S51F6000 command
the knife will go up after G0 and with G1 he goes down.
Thats the reason why your code waits on a written M3 command but there isn't 🙁
my fault...
Is it also possible to create?

Greetz Lars

fra589, 2024-02-21:
Oh! I'm afraid you will can't simply do this!  
  
The pen down / pen up is not drived by the spindle M3 / M4 / M5 command. It's driven by the G0 / G1 command:
G0 => move with pen up => no cut
G1 => move with pen down => cut

With a laser, cutting id driven by the spindle status:
M5 => laser is off => next moves will no cut
M3/M4 => laser is on => next move will cut

It may be very very difficult to insert pause between no cutting moves and cutting moves.
All moves are buffered (planned) with a timer info which define when they must be executed. This moment is calculated according to the programed speed and depending of acceleration property of the machine (very complex calculation depending of multi-axis moves, speed and acceleration).

Modify this planner algorithm seams to me very too complicated. I have neither the time nor the means to do it for you.

As @svenhb said, I don't think Lightburn is the good software for your need... Perhaps you should find a specific drag knife software.

@++;  
Gauthier.

Srallie, 2024-02-21:
Hello,

I use inkscape to create a hpgl file with knife offset included. Then i import the hpgl file in lightburn.
settings:
$30=100
$31=0

When i set the laser power for example to  50% my solenoid will reduce the knife pressure on the vinyl.

This works perfect,accept when i reduce the power,the solenoid will be take longer to go down en misses a piece of the path....

I will looking for an other program they works with M3 commands,or where i can put the G4 P## parameter.
Then i will try your piece of code again. I will report my outcomes.

greetz lars

svenhb, 2024-02-21:
>I will looking for an other program they works with M3 commands,or where i can put the G4 P## parameter.

GRBL-Plotter can do: 
PWM setup with delay: https://grbl-plotter.de/index.php?id=form-setup-3#pen-up-down
Drag tool compensation: https://grbl-plotter.de/index.php?id=form-setup-1#path-modifications

Srallie, 2024-02-21:
I will give it a try.
Thank you.

Greetz lars

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

fra589, 2024-02-23:
Hello,  
  
> Hello Sven, When i select in grbl-plotter the drag knife compensation then the program will hang.  

It's not a grbl-Mega-5X issue... You must open a new issue to grbl-plotter: https://github.com/svenhb/GRBL-Plotter/issues

> And the problem to get connected with bluetooth is still there, it didnt connect only in your program, with cable it still works  

grbl-Mega-5X doesn't have Bluetooth support... Can't be a grbl-Mega-5X issue.

@++;
Gauthier.

closed 2024-02-23 by fra589