fra589/grbl-Mega-5X #364: Driver enable delay before moving

adrianopedro, 2024-03-09:
Hello all,

I'm using a DM556T driver and apparently it needs to be enabled 200ms before any movement.
Otherwise it does some glitch in the beginning and some steps are lost.
Is there anyway to achieve this, either by commands or changing the firmware?

Thank you,
AP

fra589, 2024-03-10:
Hi @adrianopedro,  
  
You can use the parameter `$1=255` so steppers will always stay enabled.
Steppers enables pin of all axes (see https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout) will stay to 0v (In fact, they are "disable pins" signals).

@++;
Signal

adrianopedro, 2024-03-10:
Hi @fra589,

First let me thank you for your reply.
I already have it working as you suggested, but the idea was to disable the steppers after the movement to avoid having them warming up while on hold.

So I was disabling them, but when enabling I was having that effect..

I was thinking it would be possible to have some delay on the interrupt (steppers.c) to implement a delay between the enable and the pulses.

Regards,
AP
But thank you anyway

fra589, 2024-03-11:
Hi @adrianopedro,

Damn! I made a reply yesterday evening, but the reply has been lost due to a bad Internet connection :-(

It will be very dificult to add a so long delay (200ms) without break the complex algorithm of acceleration and speed control!

I think you should use the `STEP_PULSE_DELAY` compilation option, see config.h lines 567 to 576:
```
// Creates a delay between the direction pin setting and corresponding step pulse by creating
// another interrupt (Timer2 compare) to manage it. The main Grbl interrupt (Timer1 compare)
// sets the direction pins, and does not immediately set the stepper pins, as it would in
// normal operation. The Timer2 compare fires next to set the stepper pins after the step
// pulse delay time, and Timer2 overflow will complete the step pulse, except now delayed
// by the step pulse time plus the step pulse delay. (Thanks langwadt for the idea!)
// NOTE: Uncomment to enable. The recommended delay must be > 3us, and, when added with the
// user-supplied step pulse time, the total time must not exceed 127us. Reported successful
// values for certain setups have ranged from 5 to 20us.
// #define STEP_PULSE_DELAY 10 // Step pulse delay in microseconds. Default disabled.
```  
  
But the delay that you mention in your first issue post (200ms) is very too long and I think you will can't made it. `STEP_PULSE_DELAY` can work with microseconds, not milliseconds!

However, viewing the DM556T datasheet (https://www.omc-stepperonline.com/download/DM556T.pdf) I dont see the need of 200ms.
In page 11 of 13, I see:
ENA must be ahead of DIR by at least 5 microseconds
and DIR must be ahead of PUL effective edge by 5 microseconds.

So, `#define STEP_PULSE_DELAY 10` should work.

@++;
Gauthier.

closed 2024-06-21 by fra589