fra589/grbl-Mega-5X #367: alarm 11, why and what to do with it?

Krasti3049, 2024-06-01:


fra589, 2024-06-03:
**Hello**,

> alarm 11, why
From: https://github.com/fra589/grbl-Mega-5X/blob/edge/doc/csv/alarm_codes_en_US.csv

```
|  Code | Message         | Description                |
|    11 | Serial overflow | Serial RX buffer overflow. |

```
> why and what to do with it?
> _No description provided._

Without any information, It's not possible to know why and tell what to do. 

@++;
Gauthier

Krasti3049, 2024-06-03:
Arduino mega 2560 board is not original, ramps 1.4 firmware grbl 5x, program grbl plotter, lightburn, cncjs, everyone gets alarm 11

fra589, 2024-06-05:
Hi, 

It's very strange... 
You are the first to have this type of problem with all software...

Do you have made any Grbl's source code modification ?
If you made adjust in config.h or any other Grbl's file, please, post it.

Can you post the result of Grbl commands: 
`$i`
`$$`

@++;
Gauthier.

fra589, 2024-06-30:
Closed as no information for a month...

closed 2024-06-30 by fra589