fra589/grbl-Mega-5X #370: $103 parameter not working as expected.

CNCWoodworker, 2024-06-23:
Dear Gauthier et al:

Please help:

Just finished building a CNC lathe that has an X axis with a Z axis, the z Axis has a spindle with a milling tool, there is an A axis rotating along the X Axis. I will use UGS to send G-code files made with fusio. I have setup the X and Z axis correctly and have properly calibrated them. I assume the A axis steps should be configured as follows instead of steps per mm steps per one degree revolution using this formula:

(200(motor steps per revolution)*4(microstepper toshiba))/360(one revolution)*1.7(Gear Ratio for stepper to A axis), after calculating this the result is 3.7778 steps per degree. 

However, when I then issue a G0 A360 command using the UGS console command sender, the A axis spins many more revolutions than just one, and no matter what I do I get strange results. perhaps there is some configuration I'm missing, I have tried everything and have not been able to solve how to configure the $103 parameter. Also I don't know how to setup the maximum distances for a rotating axis, and the other A axis parameters.

UGS correctly recognizes the number of axis, and moves all axis, limit switches work well for homing, have not tested them during operation.

Thank you in advance for your help!

CNCWoodworker, 2024-06-23:
Forgot to mention I'm using an arduino Mega 2560 with no ramps. I wrote Fusio I meant Fusion

Thank you @fra589

fra589, 2024-06-23:
Hi @CNCWoodworker,  
  
Can you send the result of the Grbl's commands `$I` and `$$`?
Also show us the contents of your config.h file (lines 45 to 83). Those which describe the number of axes and their names.
  
You are right, $100 to $105 are travel resolution in steps per unit, so if unit are millimeters, thy are step per millimeter. If units are degree, they are step per degree. Your calculation seam good.

I assume that you have connected your axes according to the standard mapping (see https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout).
That's to say :

   | Axis number | Program index | Default axis name | Step pin | Direction pin | Disable pin |   |
   | ---: | ---: | :---: | :---: | :---: | :---: | :--- |
   | 1 | axis[0] | X | A0 | A1 | D38 | Connected to X |
   | 2 | axis[1] | Y | A6 | A7 | A2 | Not connected |
   | 3 | axis[2] | Z | D46 | D48 | A8 | Connected to Z |
   | 4 | axis[3] | A | D26 | D28 | D24 | Connected to A |

If you made another connections, perhaps you have modified the cpu_map.h file according the wiki: https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h...
If yes, send your cpu_map.h file.

Finally, I can't say it enough, every time you download a new version of Grbl to the Arduino Mega board, remember to send the flash memory reset command: `$RST=*` before sending again the $$ parameters.

@++;  
Gauthier.

CNCWoodworker, 2024-06-23:
Dear Gauthier Merci Beaucoup!

Beautiful code, brilliant!, congratulations and thank you for making it.
I reset everything and it seems to be working fine now, however, I noticed that the A axis numbers don't reset when homing, maybe that was confusing me, I'm sure that it is intentional not to reset them, but I would very much like to know how to reset everything back to zero when starting something from scratch,  maybe I should include A in the homing?, how should I accomplish that?, since I have never before worked with a rotational axis and am still understanding, I don't know. 
Also the limit switches work fine when homing but apparently not when running, is there a setup for this?

I hope I configured everything correctly.

Here is the config.h

//----------------------------------------------------------------------
// Axis definitions :
//----------------------------------------------------------------------
// IMPORTANT: When changing the axis definitions (axis numbers N_AXIS, 
// linears axis number N_AXIS_LINEAR or axes names AXIS_*_NAME, 
// don't forget to issue the reset factory defaults Grbl command: $RST=*
// if you forget the $RST=* command after change, Grbl may have 
// unpredictable behavior!
//----------------------------------------------------------------------

#define N_AXIS 5        // Number of axes (3 to 6)
#define N_AXIS_LINEAR 3 // Number of linears axis, must be <= N_AXIS

// Axis indexing and names
#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V, W, D, E & H.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'
#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'A' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'B' // Letter of axis number 5
#endif
#if N_AXIS > 5
  #define AXIS_6 5
  #define AXIS_6_NAME 'C' // Letter of axis number 6
#endif
#if N_AXIS > 6
  #error "N_AXIS must be <= 6. N_AXIS > 6 is not implemented."
#endif

Here is the $I and the $$

>>> $I
[VER:1.2h.20220109:]
[AXS:5:XYZAB]
[OPT:VNMGPH,35,255,24]
ok
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 254    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.020    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 200.000    (Homing locate feed rate, mm/min)
$25 = 2500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 12000    (Maximum spindle speed, RPM)
$31 = 550    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 80.000    (X-axis travel resolution, step/mm)
$101 = 80.000    (Y-axis travel resolution, step/mm)
$102 = 80.000    (Z-axis travel resolution, step/mm)
$103 = 3.798   
$104 = 4.000   
$110 = 5000.000    (X-axis maximum rate, mm/min)
$111 = 5000.000    (Y-axis maximum rate, mm/min)
$112 = 3000.000    (Z-axis maximum rate, mm/min)
$113 = 5000.000   
$114 = 5000.000   
$120 = 50.000    (X-axis acceleration, mm/sec^2)
$121 = 50.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$123 = 50.000   
$124 = 40.000   
$130 = 800.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
$133 = 360.000   
$134 = 360.000   

My workflow is normally to start by homing everything and using a probe to zero the part, and since I'm also learning to use a rotational axis in fusion, when it starts running the G-Code it does strange things and would love the limit switches to alarm the machine and stop instead of jamming into the physical limits.

I'm using an Arduino Giga to receive the information from UGS and send it to the Mega runing GRBL the Giga has a Display screen that shows me the positions and allows me to turn on things of the machine, such as lights etc, I did not want to mess around with GRBL it is great code, the Giga and Mega are is working fine now, here are some images from my machine.

[image]
[image]
[image]

Thank you again for the brilliant code and for your kind reply.

Best 
EZ

fra589, 2024-06-24:
Hi @CNCWoodworker,  
  
Nice machine, congratulations!

Grbl-Mega-5X treats all axes the same using their number. The names (letters) assigned to them are only used to have a standard GCode which uses the letters X, Y, Z, A, B, C, ...
Then, in config.h, we define the number of axes (`#define N_AXIS <N>` and #define `N_AXIS_LINEAR<M>` the first <M> axes will be processed with linear measurements and the conversion between millimeters and inches will be carried out according to GCode G21/G20 Axes whose number is between <M> and <N> will not be modified by G21/G20.

In the Grbl-Mega-5X default configuration, only the first 3 axes are homed and their coordinates reset.
If you need to reset also your A axis, you need to adjust the homing configuration in config.h. It's pretty simple: 
Edit the config.h file and, for your special use case, change from line 247 to 253 as this:

```
#elif N_AXIS == 5 // 5 axis : homing
  #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis
  #define HOMING_CYCLE_1 (1<<AXIS_1) // Home X axis
  #define HOMING_CYCLE_2 (1<<AXIS_4) // Home A axis
```

In the homing sequence, you need to suppress the Y axis homing, otherwise, the homing sequence will block, waiting for the missing axis.

Regarding hardware limits outside of the homing function, this functionality is not activated by default in Grbl-Mega-5X due to a small but real performance impact.
To enable this functionality, you need to edit the cpu_map.h file and uncomment the line number 206 like this:  
`#define ENABLE_RAMPS_HW_LIMITS`  
Then, set the Grpl running parameter `$21=1`.

With hardware limits, when a limit switch is triggered, the axes are immediately stopped without deceleration causing loss of position (possible steps lost on steppers motors).

I advise instead to use logical limits in Grbl, with a good configuration of homing and the max axes travels parameters ($130 to $135). This is more efficient and when the soft limit is reached, the machine will stop without lost it's position.
In your case, after correctly configured $130, $132 and $133, set $20=1 to enable soft limits.

After the config.h and cpu_map.h changes, you will need to recompile and upload Grbl-Mega-5X.
See the wiki procedure, and if you use the Arduino IDE to do this, directly edit the files inside the library folder or don't forget to delete the prior library to avoid compiling and upload issue.

@++,
Gauthier.

CNCWoodworker, 2024-06-24:
Hi Gauthier:Thank you, I will do it with the soft limits, it is a great idea!Thank you for your reply and the informationBest.EZSent from my iPhone

Then, in config.h, we define the number of axes (#define N_AXIS <N> and #define N_AXIS_LINEAR<M> the first  axes will be processed with linear measurements and the conversion between millimeters and inches will be carried out according to GCode G21/G20 Axes whose number is between  and  will not be modified by G21/G20.

CNCWoodworker, 2024-06-25:
Bonjour Gauthier!

I configured and activated the soft limits, I understand  why you recommended it is much better than jamming into the limits and leaving the machine in an unknown state, thank you! 
Now I'm struggling with Fusion, it appears that the post processors they have for GRBL either use only 3 axis or 6, so when the toolpaths are being calculated it generates paths for the Y, and B and C axis, which my machine does not have, I'm trying to modify the post processor to accomodate my strange and peculiar machine. It has been many years since I wrote code (40+) back then turbo pascal was the thing, and I have no experience in Javascript, I guess I'll have to learn. 
Do you know of anyone that has made a post processor for GRBL 5x?

fra589, 2024-06-25:
Hello @CNCWoodworker!  
  
I'm sorry, but being a fervent defender of free software, I do not use and do not know Fusion well enough to answer your question.
I use only FreeCAD for the generation of my GCodes. Moreover, I actively participated in writing the post-processor for Grbl and it is perfectly compatible with grbl-Mega-5X.

Perhaps you can test renaming your axes from X, Z and A to Y, B and C in the config.h grbl's file for making grbl-Mega-5X compatible with Fusion post-processor output?

@++;
Gauthier.

CNCWoodworker, 2024-06-25:
Thank you Gauthier!

I will definitely try FreeCAD. I love the free software movement, and open source.

Best
EZ

closed 2024-06-30 by fra589