fra589/grbl-Mega-5X #371: Code error in motion_control.c?

ccwtruck, 2024-06-30:
Hello, Gauthier.

Well, I finally got my limit switches installed and configured, enabled homing, and tried to home the machine.  I found out that I had a couple of misconfigurations in config.h, so I corrected those and tried to compile/ upload a new copy of Mega 5X.

In the Arduimo IDE (Debian), I am now getting a compile error:
```
/home/john/Arduino/libraries/grbl/motion_control.c: In function ‘mc_homing_cycle’:
/home/john/Arduino/libraries/grbl/motion_control.c:422:36: error: expected ‘;’ before ‘)’ token
       limits_go_home(HOMING_CYCLE_1);  // Homing cycle 1
                                    ^
/home/john/Arduino/libraries/grbl/motion_control.c:422:36: error: expected statement before ‘)’ token
exit status 1

```
that I have never gotten before.  

I changed nothing outside of config.h, and all I changed were the order of my homing cycles, enabled HOMING_FORCE_SET_ORIGIN to put the table into positive space, and set N_HOMING_LOCATE_CYCLE  to 3 since I have defined 3 homing cycles.  

My homing cycles are defined as follows:
```
  #define HOMING_CYCLE_0 (1<<AXIS_3)   // Home Z axis first to clear workspace.
  #define HOMING_CYCLE_1 (1<<AXIS_1))  // Home X axis
  #define HOMING_CYCLE_2 ((1<<AXIS_2)|(1<<AXIS_4)) // Home Y & cloned Y axis together  

```
I do not see anything that would trigger this compile error.  Do you have any idea what could have caused it to crop up now?

John

fra589, 2024-07-01:
Hi John,  
  
Your error is not in motion_control.c, but really in config.h. 
In your line:
`#define HOMING_CYCLE_1 (1<<AXIS_1))  // Home X axis`
You have two closing parenthesis ")" instead of one!

In addition, the N_HOMING_LOCATE_CYCLE definition is not used to change the number of HOMING_CYCLE_* definitions but the number of times that all homing cycles will be performed.

So, you can very well define:
```
 #define HOMING_CYCLE_0 (1<<AXIS_3)   // Home Z axis first to clear workspace.
  #define HOMING_CYCLE_1 (1<<AXIS_1)  // Home X axis
  #define HOMING_CYCLE_2 ((1<<AXIS_2)|(1<<AXIS_4)) // Home Y & cloned Y axis together
```

And let:
`#define N_HOMING_LOCATE_CYCLE 1 // Integer (1-128)`

@++;
Gauthier

ccwtruck, 2024-07-01:
@fra589 ,

Ah-ha! I see that now.

It's actually a moot point, though; I have gone back to gnea/grbl-Mega with the Y and A axis drivers slaved.  It homed perfectly last night, and the X gantry stays square.

One problem with Mega-5x that I could never quite iron out was that occasionally and randomly, the X gantry would get out of square and I would have to shut down and manually turn the ball screws to square it.  When this would happen, you could tell that one motor was not being driven.  Trying to jog the Y axis the opposite way in hopes of returning it to square rarely worked. 

Another issue was that it seemed to not want to act on a triggered limit switch and stop movement -- several times, the metal vane that actuated the proximity switch was crushed when it crashed into the switch.

All in all, for my purposes, it was just not reliable enough to use for production in my shop.  So I changed back to gnea/grbl-Mega and reliability was restored -- the X gantry stayed square, the axises stopped when a limit switch was tripped, and home worked reliably every time.

Thank you for your quick responses to every issue, but I just don't see Mega-5x being the solution for me.  Sorry!

John

closed 2024-07-01 by ccwtruck