fra589/grbl-Mega-5X #374: Soft limits will not allow A Axis movement in any positive direction

CNCWoodworker, 2024-07-21:
Bon Jour Gauthier #fra589: 

How are you? I'm almost ready setting up my machine to work with soft limits as you recommended. I would like some verification that limits are not exceeded for machine safety, and as you suggested enabling hard limit verification while operating does affect performance. When I enable the soft limits, my A axis is only allowed to move in a negative direction. if the A is starts at a negative value it does allow movement back to zero, this behaviour seems correct for linear axis, but I think that for a rotary unlimited moving axis this is incorrect. 
Is there a setting I should use? or any recommendation?

fra589, 2024-07-24:
Hi @CNCWoodworker,
  
Sorry for the late reply, I'm traveling very far from home.
For your unlimited A rotational axis, you can use 0 (zero) as axis maximum travel, this will disable the soft limit for this axis. So, it will be unlimited as you need.
See https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration#130-131-132-133-134-135--xyzabc-max-travel-mm-degree

@++;
Gauthier.

CNCWoodworker, 2024-07-24:
Merci Beaucoup!Bon VoyageEZSent from my iPhone