fra589/grbl-Mega-5X #375: mega 2560 5x  Stepper motor does not move

parkgeonhun, 2024-07-27:
When you enter the program, you can hear the sound of the stepper motor running, but the motor actually vibrates and does not move. And the a and b axes do not work. It also says that there is an error in the Arduino code.
[image]
[image]

fra589, 2024-07-27:
Hi @parkgeonhun,

> the stepper motor running, but the motor actually vibrates and does not move.

Probably an Hardware and/or config problem
- too load on the motor,
- speed or acceleration or both too high,
- ...

What is your config ? 
- `$I` and `$$` results,
- Any change in config.h, default.h & cpu_map.h ?

> there is an error in the Arduino code

Do you read and respect the procedure : https://github.com/fra589/grbl-Mega-5X/wiki/Compiling-grbl-Mega-5X ?

@++;
Gauthier.

parkgeonhun, 2024-07-29:
What version of Arduino version does it use? If it's the latest version, I need a zip file. I just can't upload the file.

parkgeonhun, 2024-07-29:
Also, how do I resolve this error?
[image]

closed 2024-07-30 by fra589