fra589/grbl-Mega-5X #378: Extra digital outputs?

DDanielMK, 2024-08-06:
Good day
Firstly, this is a very cool project. Thanks for putting in all of the effort and making it freely available @fra589. 

I am using Mega5X to control a 5-axis 3D printing apparatus for my research project. I have 13 outputs that I need to be able to turn on/off individually. Previously when I was operating the machine in 3-axis mode with Marlin, I used 13 of the Ramps pins that are meant for the screen as digital outputs, and used M42 to toggle between 0 and 5V. 

I have now got the four available digital output pins working with Mega5X using M62/63. Is it at all possible to add extra digital outputs? As far as I can see there are still unused pins on the Ramps board, alongside the digital IO pins that have already been allocated. 

This would avoid me having to connect up a second Arduino to the Ramps and use the existing four outputs from the Ramps to tell the second Arduino what to do, which will be a bit of a complicated and roundabout way of doing it.  

I would appreciate any suggestions. 

Thanks!

fra589, 2024-08-07:
Hi @DDanielMK, 

It should be not too difficult to do...
The output pins are defined in cpu_map.h, from line 224 to 237. You can add your needed pins here. Each pin is defined with it's DDR, PORT ans BIT. See the Arduino doc (https://docs.arduino.cc/retired/hacking/hardware/PinMapping2560/) and the cpu_map.h mapping definition in the wiki (https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h) to define your own supplementary pins.

Then, in digital_control.c:
- change the type of function:
`uint8_t digital_get_state()` to `uint16_t digital_get_state()`
- and the variable at line 56:
`uint8_t digital_state = DIGITAL_OUTPUT_STATE_OFF;`to `uint16_t digital_state = DIGITAL_OUTPUT_STATE_OFF;`
- Add the lines to drive your new output pins:
- in the function digital_init() after the line 30,
- in the function digital_get_state() after the line 85 (7 lines to add for each new pins, copy and increment the already defined pins)
- change the function `void digital_stop(const uint8_t mode)` to `void digital_stop(const uint16_t mode)` 
- and add your new pins after line 153,
- change the function `void digital_set_state(uint8_t mode)` to `void digital_set_state(uint16_t mode)`,
- and add your new pins after the line 214
- change the function `void digital_sync(uint8_t mode)` to `void digital_sync(uint16_t mode)`

In the gcode.c (line 30) file, change the max number of your digital output to `#define MAX_DIGITAL_OUTPUT 13` (the maximum you will can put here will be 16),
Then, from line 1557 to line 1573, change all the `uint8_t` types to `uint16_t`.

This should be work...

Please, report if it work for you.

@++;
Gauthier.

DDanielMK, 2024-08-07:
Thank you @fra589 for the detailed instructions. Those steps worked, except I also had to define the extra outputs in ditigal_control.h, and change all instances of uint8_t to uint16_t in the same file. 

I am using digital pins 16, 17, 23, 25, 27, 29, 31, 33, 35, 37, 39, 41, 43 and 45. I don't need to use 27 to 33 as inputs, so I re-defined them as outputs instead. 

The only thing that is puzzling me now is how to invert the digital output pins so I don't have to re-wire my relays. I have defined INVERT_DIGITAL_OUTPUT_PIN_X for all the output pins in config.h. I have also added the necessary code for each pin in the _void digital_stop_ and _void digital_set_state_ functions in digital_control.c. Only the first four pins are inverted and the rest stay like they were. Is there a setting that I am missing somewhere? 

Thanks for the help :)

DDanielMK, 2024-08-08:
Thanks @fra589 

I have attached the files. 

[Mega5x.zip](https://github.com/user-attachments/files/16541461/Mega5x.zip)

fra589, 2024-08-08:
I don't see anything wrong, except that you have 14 outputs and not 13. :smile: 

Can you also send your cpu_map.h and gcode.c so I will can make complete test on my development board?

@++;
Gauthier.

DDanielMK, 2024-08-12:
Here are the other two files. I'll change the number of outputs, thanks. 

[CPUmap and Gcode.zip](https://github.com/user-attachments/files/16578012/CPUmap.and.Gcode.zip)

fra589, 2024-08-12:
Hi @DDanielMK ,  
  
I found the bug! :smile: 
In fact, your extra pins were working in reverse as you want, but it was their initialization that wasn't working.

In the function digital_init() in the file digital_control.c, there is a small line (number 41) which call `digital_stop(0x0F);`
And the mode 0x0F specify only the first 4 bits! => So the first 4 outputs.
Change the ilne 42 to:
`digital_stop(0xFFFF);`
And grbl-Mega-5X will work as you want.

@++;
Gauthier.
P.S. I send to you the modified file report.c where I added the report for all your digital outputs.
[report.c.zip](https://github.com/user-attachments/files/16589087/report.c.zip)

DDanielMK, 2024-08-15:
Thank you @fra589 :-)

closed 2024-08-19 by fra589