fra589/grbl-Mega-5X #380: Hard limit and homing ramps don´t working

xero83, 2024-11-12:
Hello, I have installed the latest version of GRBL and am using the RAMPS 1.4. My configuration is as follows: I have uncommented the line #define ENABLE_RAMPS_HW_LIMITS, $20=0, $21=1, $22=1. I have also checked that the endstops are functioning correctly as normally open (NO) and that, when activated, they show as "Triggered" in the terminal.

With this configuration, whether the machine is running a program or in standby, activating any endstop neither stops the machine nor triggers any alarm or anything else. Due to the characteristics of the machine, I need to perform homing with the physical endstops and the limit alarms. From what I’ve seen, it should be functional, but it’s not working for me, and I’ve tried everything I can think of. I would appreciate any kind of help.

fra589, 2024-11-12:
Hi @xero83,

Hum... This strange! With `#define ENABLE_RAMPS_HW_LIMITS` and `$21=1`, hard limits should work.
Can you join your config.h and cpu_map.h file and post the result of `$I` and `$$` commands please ?

Hard limits and Homing use different pieces of code and are independent.
Do you confirm that the Homing doesn't work too ?

@++
Gauthier.

xero83, 2024-11-12:
Tomorrow first thing in the morning, I'll send you what you're asking for, as I have the machine at work.

Freytag85, 2025-11-21:
Found the Problem. My bad. I tried it with simulating the endstop by trigger it manually. Problem was that the machine was not moving on this axis. If it‘s moving and you trigger it, it works fine.