fra589/grbl-Mega-5X #381: Laser issue

bissicletta, 2024-11-26:
_**_Hello! How are you? I'm finishing my CNC and I want to put the laser on it but I'm having problems, I'm using ramps 1.4
and a laser lt-40W-aa from laser tree, in the connections description, it says PWM )0/3-12v, 0-5khz.
What I see is that it handles a pwm between 6 and 12v, so I connected D9 to the pwm, D9+ to 12v and negative to gnd
(to the input D driver adapter P-da-01) I also tried all the possible configurations, commenting or uncommenting invert, separate, output, I tried in D8, I supported pwm in lasergrbl, $RST=* , $30=255 $31=1 $30=1000_**_

**_The problem is that the laser is always on, m5 turns it on, m3 or m4 turns it off (sxxx does not work) I'm already thinking that it could be a fault in the mosfet or that the khz or rpm are incorrectly adjusted, something like that._**

```

// Spindle, laser and other PWM output
//----------------------------------------------------------------------
// Chose the spindle pin output :
// SPINDLE_PWM_ON_D8  => 0-12v 16 bits PWM on RAMPS D8 (default)
// SPINDLE_PWM_ON_D9  => 0-12v 8 bits PWM on RAMPS D9
// SPINDLE_PWM_ON_D6  => 0-5v 8bits PWM on RAMPS Servo 2 signal (Mega 2560 D6)
// Uncomment the line which correspond to your hardware
//#define SPINDLE_PWM_ON_D8
#define SPINDLE_PWM_ON_D6
//#define SPINDLE_PWM_ON_D9

// Spindle PWM signal inversion:
// In case of particular electronics, it may be necessary to invert the values
// of the PWM signal of the spindle. For example, if the minimum spindle 
// rpm is 1 and maximum is 1000, M3S250 will output 75% instead of 25% and
// M3S750 will output 25% instead of 75%. Disabled by default
//#define INVERT_SPINDLE_PWM_VALUES

// Use different spindle output pin in laser mode:
// Spindle or laser tools do not have the same hardware specifications.
// When using both spindle and laser on the same machine it will be useful
// to have spindle and laser on diffrents pins which can deliver the
// differents outputs nedded.
//----------------------------------------------------------------------
// ! IMPORTANT: When changing the SEPARATE_SPINDLE_LASER_PIN compil option,
// don't forget to issue the reset factory defaults Grbl command: $RST=*
// if you forget the $RST=* command after change, Grbl may have 
// unpredictable behavior!
//----------------------------------------------------------------------
// Uncomment the next line to enable this functionality (default disabled):
#define SEPARATE_SPINDLE_LASER_PIN

#ifdef SEPARATE_SPINDLE_LASER_PIN
  // Laser PWM can be on D6 (default) or on D8 or D9. 
  //#define LASER_PWM_ON_D6
  //#define LASER_PWM_ON_D8
  #define LASER_PWM_ON_D9
#endif

// Use output PWM drived by GCode command M67(Analog Output,Synchronized) 
// or GCode command M68(Analog Output, Immediate).
//----------------------------------------------------------------------
// ! IMPORTANT: When changing the USE_OUTPUT_PWM compil option,
// don't forget to issue the reset factory defaults Grbl command: $RST=*
// if you forget the $RST=* command after change, Grbl may have 
// unpredictable behavior!
//----------------------------------------------------------------------
// Uncomment the next line to enable the use of M67/M68 PWM output (Disabled by default).
//#define USE_OUTPUT_PWM

#ifdef USE_OUTPUT_PWM
  // Optional PWM can be on D9 (default) or on D8 or D6. 
  // Warning ! Optional can't use the same timer than the spindle or the laser pin
  // For more information about this, see the comment of the relevant section in cpu_map.h 
  //#define OUTPUT_PWM_ON_D9
  //#define OUTPUT_PWM_ON_D8
  //#define OUTPUT_PWM_ON_D6
```

**_cpu_map it`s default_** 

 #elif defined (LASER_PWM_ON_D9)

      // Set Timer up to use TIMER2B which is attached to Digital Pin 9 - Ramps D9
      #define LASER_PWM_MAX_VALUE     255.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
      #ifndef LASER_PWM_MIN_VALUE
        #define LASER_PWM_MIN_VALUE   1   // Must be greater than zero.
      #endif
      #define LASER_PWM_OFF_VALUE     0
      #define LASER_PWM_RANGE         (LASER_PWM_MAX_VALUE-LASER_PWM_MIN_VALUE)

      //Control Digital Pin 9 
      #define LASER_TCCRA_REGISTER    TCCR2A
      #define LASER_TCCRB_REGISTER    TCCR2B
      #define LASER_OCR_REGISTER      OCR2B
      #define LASER_COMB_BIT          COM2B1

      // 1/8 Prescaler, 8-bit Fast PWM mode
      #define LASER_TCCRA_INIT_MASK ((1<<WGM20) | (1<<WGM21))
      #define LASER_TCCRB_INIT_MASK ((1<<WGM22) | (1<<CS22))
      #define LASER_OCRA_REGISTER   OCR2A // 8-bit Fast PWM mode requires top reset value stored here.
      #define LASER_OCRA_TOP_VALUE  0xFF  // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

      // Define spindle LASER pins.
      #define LASER_PWM_DDR   DDRH
      #define LASER_PWM_PORT  PORTH
      #define LASER_PWM_BIT   6 // MEGA2560 Digital Pin 9

fra589, 2024-11-26:
Hi @bissicletta,  
  
When you use the `#define SEPARATE_SPINDLE_LASER_PIN` option, the $32 (laser mode) parameter intervenes to select the destination of the M3, M4 & M5 commands.
- if `$32 = 0` => the output is directed to the spindle (D6 in the configuration you posted).
- if `$32 = 1` => the output is directed to the laser (D9 in your configuration).
In this operating mode, M3 turns on the laser taking into account the speed (the power SXXX is regulated according to the acceleration of the movements). M4 turns on the laser at the requested power SXXX without taking into account the acceleration.
Finally, M5 turn off the laser. For safety reasons (possibility of fire), the laser turns off if the speed is zero.

@++;
Gauthier.

bissicletta, 2024-11-26:
That's the problem, I put m5 and the laser turns on

and putting any of these, the laser turns off m4s1 m4s0 m4s255 m4s1000 / m3s0 m3s1 m3s255 m3s1000

even putting $32=1 or $32=0 what changes is that I can turn off the laser, with $32=0 it only stays on

fra589, 2024-11-27:
!?!?!?
Can you post a zip file with the grbl-Mega-5X sources you use ?
(Only the `grbl` directory with all the `.c` and `.h` files)

bissicletta, 2024-11-27:
I am attaching a zip
[grbl.zip](https://github.com/user-attachments/files/17934219/grbl.zip)
 of the grbl folder, so it is working as I described, I made some modifications to test but it continues the same! Thanks!

bissicletta, 2024-12-02:
**Sorry, I failed so many times that I started googling and tried anything!
I just downloaded the original code again
The only thing I changed was that the Dir axis and step of the z axis:**

 ```
 #define STEP_PORT_0 F
  #define STEP_PORT_1 F
  #define STEP_PORT_2 F
  #if N_AXIS > 3
    #define STEP_PORT_3 A // Axis number 4 (Ramps E0)
  #endif
  #if N_AXIS > 4
    #define STEP_PORT_4 C // Axis number 5 (Ramps E1)
  #endif
  #if N_AXIS > 5
    #define STEP_PORT_5 L // Axis number 6 (Ramps Aux-3 D49)
  #endif
  #define STEP_BIT_0 0  // X Step - Pin A0
  #define STEP_BIT_1 6  // Y Step - Pin A6
  #define STEP_BIT_2 3  // Z Step - Pin D3
  #if N_AXIS > 3
    #define STEP_BIT_3 4 // Axis number 4 Step - Pin D26
  #endif
  #if N_AXIS > 4
    #define STEP_BIT_4 1 // Axis number 5 Step - Pin D36
  #endif
  #if N_AXIS > 5
    #define STEP_BIT_5 0 // Axis number 6 Step - Pin D49
  #endif
  #define _STEP_BIT(i) STEP_BIT_##i
  #define STEP_BIT(i) _STEP_BIT(i)
  #define STEP_DDR(i) _DDR(STEP_PORT_##i)
  #define _STEP_PORT(i) _PORT(STEP_PORT_##i)
  #define STEP_PORT(i) _STEP_PORT(i)
  #define STEP_PIN(i) _PIN(STEP_PORT_##i)

  // Define step direction output pins.
  #define DIRECTION_PORT_0 F
  #define DIRECTION_PORT_1 F
  #define DIRECTION_PORT_2 F
  #if N_AXIS > 3
    #define DIRECTION_PORT_3 A // Axis number 4 (Ramps E0)
  #endif
  #if N_AXIS > 4
    #define DIRECTION_PORT_4 C // Axis number 5 (Ramps E1)
  #endif
  #if N_AXIS > 5
    #define DIRECTION_PORT_5 B // Axis number 6 (Ramps Aux-3 D51)
  #endif
  #define DIRECTION_BIT_0 1 // X Dir - Pin A1
  #define DIRECTION_BIT_1 7 // Y Dir - Pin A7
  #define DIRECTION_BIT_2 4 // Z Dir - Pin A4
```

**and in config, just this:**
```
// Chose the spindle pin output :
// SPINDLE_PWM_ON_D8  => 0-12v 16 bits PWM on RAMPS D8 (default)
// SPINDLE_PWM_ON_D9  => 0-12v 8 bits PWM on RAMPS D9
// SPINDLE_PWM_ON_D6  => 0-5v 8bits PWM on RAMPS Servo 2 signal (Mega 2560 D6)
// Uncomment the line which correspond to your hardware
#define SPINDLE_PWM_ON_D8
//#define SPINDLE_PWM_ON_D6
//#define SPINDLE_PWM_ON_D9

// Spindle PWM signal inversion:
// In case of particular electronics, it may be necessary to invert the values
// of the PWM signal of the spindle. For example, if the minimum spindle 
// rpm is 1 and maximum is 1000, M3S250 will output 75% instead of 25% and
// M3S750 will output 25% instead of 75%. Disabled by default
//#define INVERT_SPINDLE_PWM_VALUES

// Use different spindle output pin in laser mode:
// Spindle or laser tools do not have the same hardware specifications.
// When using both spindle and laser on the same machine it will be useful
// to have spindle and laser on diffrents pins which can deliver the
// differents outputs nedded.
//----------------------------------------------------------------------
// ! IMPORTANT: When changing the SEPARATE_SPINDLE_LASER_PIN compil option,
// don't forget to issue the reset factory defaults Grbl command: $RST=*
// if you forget the $RST=* command after change, Grbl may have 
// unpredictable behavior!
//----------------------------------------------------------------------
// Uncomment the next line to enable this functionality (default disabled):
#define SEPARATE_SPINDLE_LASER_PIN

#ifdef SEPARATE_SPINDLE_LASER_PIN
  // Laser PWM can be on D6 (default) or on D8 or D9. 
  //#define LASER_PWM_ON_D6
  //#define LASER_PWM_ON_D8
  #define LASER_PWM_ON_D9
#endif
```

**but it remains exactly the same! I also attached a picture of the laser, just in case I'm doing something wrong and it's not the code.**

[image]

fra589, 2024-12-03:
Hi @bissicletta

I think you don't have the right connection...
As I see in the laser documentation here (https://cdn.shopify.com/s/files/1/0549/7500/4823/files/LT-40W-AA_User_Manual.pdf) should be like this:
[image]

@++;
Gauthier.

bissicletta, 2024-12-04:
Unfortunately now it is always on without responding to any command, I tried with pwm output as well but I didn't see any changes, I'm going to check the ramps 1.4 mosfet and update information!

Thank you very much for your time and concern!!

bissicletta, 2025-01-02:
Hello again! First of all, have a great 2025!
I had a breakthrough, I connected d9+ d9- and gnd directly to the laser
and I realized that when I changed the frequency to 0 in grbl laser ($33=0)
it works correctly, m5 turns off the laser
and m4 or m3 turn it on
but the power does not vary, it does not respond to the "sxxx" command

the mosfets are fine, and I also tried changing to d8 d6 and it's still the same! any ideas what it could be?

fra589, 2025-01-02:
Hi @bissicletta, Happy new year to you too!

!?!?!? $33 is not a frequency parameter. It's the X axis maximum speed parameter!

Perhaps you don't have read the full laser mode documentation:
https://github.com/fra589/grbl-Mega-5X/wiki/Laser-Mode

Laser power is movement dependent and the power is on only in G1 moves mode. Not in G0 (rapid move) mode.

How do you made the laser power measurement?

@++;
Gauthier.

closed 2025-11-21 by fra589