fra589/grbl-Mega-5X #382: M62&63 causing pause in toolpath exectution

DDanielMK, 2025-01-10:
Hi there

I have successfully set up Mega-5X on a 5-axis 3D printing setup, and I am using the M62 and M63 to toggle the states of a number of relays. I guess this is to be expected, but whenever I use M62 or M63, there is a brief pause in the motion of the axes while the relays are switched. This isn't a major issue as the axes are moving quite slowly, but it would be nice if the M commands could be executed without the axes having to stop. I am fairly sure that I wasn't having these delays when I was using M42 on Marlin to do the same thing on a 3-axis setup. 

I am not very knowledgeable when it comes to what is happening in the registers etc, but if there is a way to get rid of the delay that would be great. 

Thanks :)

D

fra589, 2025-01-10:
Hi @DDanielMK,  
  
The management of the digital output features of grbl-Mega-5X uses the same standard as LinuxCNC.
http://linuxcnc.org/docs/stable/html/gcode/m-code.html#mcode:m62-m65
The M62 and M63 commands are synchronized with the movement. This means that grbl-Mega-5X waits for the end of the current buffered movement (end of movements + deceleration + stop) before changing the output state of the pin, then it resumes the rest of the movements (acceleration then stabilization of the speed).
The brief pause that you observe in the movement is therefore normal.
The M64 and M65 commands (turn on/off digital output immediately) work independently of the movement and have no impact on it.
But be careful, when using M64 and M65, the digital outputs change state immediately, without taking into account the upcoming movements that are already in the buffer.  
  
I don't know enough about Marlin to know if the M42 command is synchronized or not.  
  
@++;  
Gauthier.

DDanielMK, 2025-01-14:
Thanks for the clarification @fra589 

If I use M64/65 then the outputs are triggered before the correct point on the toolpath, which is to be expected according to your explanation above. 

Is there any way of switching the outputs at a particular point on the toolpath without affecting the movement? Otherwise I will just use M62/63.

Thanks again :)
D

fra589, 2025-01-14:
Hi @DDanielMK,  
  
> If I use M64/65 then the outputs are triggered before the correct point on the toolpath, which is to be expected according to your explanation above.

Yes, you are right.

> Is there any way of switching the outputs at a particular point on the toolpath without affecting the movement? Otherwise I will just use M62/63.

Unfortunately for you, it's not possible.  
The GCode interpreter code calculates the geometric elements of the movement, then sends these elements to another module which calculates the temporal elements according to the requested speed and acceleration constraints. It is thus translated into motor pulses which are sent to the buffer controlled by the timer.
Several GCode movement orders are therefore stacked in the buffer and only the processor timer routine is responsible for sending them to the motors at the right time.
According to this, the only way for the GCode interpreter to know the state of the movement is to wait for it to be completed.

@++;
Gauthier.

DDanielMK, 2025-03-21:
Thanks @fra589 
Fortunately the pause when using M62 or M63 is not too much of an issue for me at the moment, as my axes are moving quite slowly. 

D

closed 2025-03-21 by DDanielMK