fra589/grbl-Mega-5X #383: Using Ramps + 6'th stepper motor?

DDanielMK, 2025-03-21:
Greetings again :)

I would like to be able to add a 6'th axis to my system. I have seen this thread about removing the Ramps and using separate motor drivers (as far as I understand it): https://github.com/fra589/grbl-Mega-5X/issues/26

Is it possible to use Ramps with one extra driver for the 6'th axis, and then to re-allocate the necessary pins so that the extra driver can be wired from the Ramps? 

Thanks 
D

fra589, 2025-03-21:
Hi @DDanielMK,

Yes, it's fully possible.

This what I did to test the program when I created it, using a "keyes stepper" driver (https://www.google.com/search?q=keyes+stepper) to use the same driver than other motors.
According with the pinout wiki (https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout), you can connect the external driver to the RAMP Aux-3 pins:
- Step pin on D49
- Direction pin on D51
- Disable pin on D53
cf. https://reprap.org/mediawiki/images/c/ca/Arduinomega1-4connectors.png

You also can change those hardware mapping in cpu_map.h after reading this wiki: https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h

@++;
Gauthier

DDanielMK, 2025-03-21:
Great, thanks @fra589. 

:D