fra589/grbl-Mega-5X #384: Homing with cloned axis and limit switches

Rigorini, 2025-03-22:
I am configuring a 3-axis CNC machine using 5 motors: 2 X, 2 Y, and 1 Z. I cloned X and Y and reversed the rotation of one Y motor using the command $3=00010000.

Below is the axis configuration:

#define N_AXIS 5        // Number of axes (3 to 6)
#define N_AXIS_LINEAR 3 // Number of linear axes, must be <= N_AXIS

// Axis indexing and names
#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V, W, D, E & H.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'
#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'X' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'Y' // Letter of axis number 5
#endif
#if N_AXIS > 5
  #define AXIS_6 5
  #define AXIS_6_NAME 'C' // Letter of axis number 6
#endif
#if N_AXIS > 6
  #error "N_AXIS must be <= 6. N_AXIS > 6 is not implemented."
#endif

I have set up the homing as follows:

#elif N_AXIS == 5 // 5 axis: homing
  #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
  #define HOMING_CYCLE_2 ((1<<AXIS_1)|(1<<AXIS_4)) // Home X axis.
  #define HOMING_CYCLE_1 ((1<<AXIS_2)|(1<<AXIS_5)) // Home Y axis.

When I send the $h command, the motors rotate correctly, but the problem is that when I press the limit switch, only one of the two motors on an axis stops.
For example, AXIS_2 stops, but its clone, AXIS_5, does not. 

How can I solve it?

fra589, 2025-03-24:
Hi @Rigorini,

Sorry for my late reply, as all retired person, I'm always hardly busy :-D

The behavior you're experiencing is right.

By default, grbl-Mega-5X use a limit switch on each axis. This permit automatically adjusts the machine's squaring.

If you don't want to use two limit switches for cloned axes, you need to edit the cpu_map.h file (lines 139 to 198) and set the same values ​​for:
- MIN_LIMIT_PORT_0 and MIN_LIMIT_PORT_3,
- MIN_LIMIT_BIT_0 and MIN_LIMIT_BIT_3,
- MAX_LIMIT_PORT_0 and MIN_LIMIT_PORT_3,
- MAX_LIMIT_BIT_0 and MIN_LIMIT_BIT_3,
- MIN_LIMIT_PORT_1 and MIN_LIMIT_PORT_4,
- MIN_LIMIT_BIT_1 and MIN_LIMIT_BIT_4,
- MAX_LIMIT_PORT_1 and MIN_LIMIT_PORT_4,
- MAX_LIMIT_BIT_1 and MIN_LIMIT_BIT_4,
In this way, both cloned X axes will use the same switch. limit switch and the two cloned Y axes as well.

For more information on the structure of the cpu_map.h file, see the wiki: https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h

@++;
Gauthier.

Rigorini, 2025-03-24:
It works! Thank you very much

closed 2025-03-24 by Rigorini