fra589/grbl-Mega-5X #385: Using A axis driver pin out as C axis in grbl-mega-5x and mks gen-l v1.0 control board

Horan31, 2025-05-16:
Hello 

[image]

I'm building a new desktop 5axes cnc milling machine and i'm using an mks genl v1.0 board and grbl-mega-5x firmware, it's architecture is X ,Z and B (rotation over X axis)axes for the head ,Yand C(rotation over Z axis)axes for the table , i want to use the driver of A axis (E0 in mks genl v1) for the C axis ,is there any possibility to configure the firmware and replace the pins.
I've attached the pin mapping of the mks genl v1 have a look.
Thanks,

fra589, 2025-05-16:
hi @Horan31,

I don't know well the mks board. I don't know on which pins the axes are connected.
But grbl-Mega-5X is very flexible in its configuration.
Internally, grbl-Mega-5X use axis numbers from AXIS_1 to AXIS_N (with N between 3 and 6). The GCode axes number and names definition is defined in the config.h file (line 55 to 82).
Simply affect the letter you want to the axis number you want, then recompile grbl-Mega-5X.
Don't forget to issue the grbl's "$RST=*" commande after each axes definition change.

@++;
Gauthier

Horan31, 2025-05-16:
@fra589 that's for replying,
- The mks board is a compact All in one mother board designed for 3d printers and it's compatible with grbl-mega-5x because it's an Arduino atmega2560 integrated with Ramps 1.4 shield so the pin mapping is the same as Ramps 1.4 shield.
- and through what you said my config.h  must be edited like this 👇: 
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'C' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'B' // Letter of axis number 5
#endif
#if N_AXIS > 5
  #define AXIS_6 5
  #define AXIS_6_NAME 'A' // Letter of axis number 6
#endif
#if N_AXIS > 6
  #error "N_AXIS must be <= 6. N_AXIS > 6 is not implemented."
#endif
For cpu.h or any other file nothing to be changed ? If yes, what change i have to do .

fra589, 2025-05-16:
@Horan31, 
Yes, the config.h change you wrote should work as you want.
You don't have to change any other file as the mks board have the same pin mapping than the RAMP board.

@++;
Gauthier.

Horan31, 2025-05-17:
@fra589 
Okay thank you , i will try this the next day and see it works .
Other question i want to add an emrgency stop button that disable the motion and stops everything from moving .is  the grbl-mega-5x programmed to have one or i need to build another circuit to cut the power off when stop button is pressed?!

fra589, 2025-05-17:
Good morning @Horan31,

On my own CNC milling machine, I use the reset pin for emergency stop button. It work well for me.
As I wrote in the wiki (https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout), the reset switch pin is A9 (pin in Aux-2 on RAMPS board).
When the reset pin is connected to the ground, all movements are stopped instantly and grbl-Mega-5X will reset.

@++;
Gauthier.

Horan31, 2025-05-17:
Good afternoon @fra589, 
Okay i will try this,
Sorry for too many questions but i have to because i need to finish assebling the machine .
Another question if you don't mind ,i want to use the Xmax and Ymax endstops pins for the Bmin and Cmin endstops for both B and C axes how to configure grbl-mega-5x.

fra589, 2025-05-18:
Good morning @Horan31,  

To change the limit pin assignment, you need to make some changes in the cpu_map.h file.
The pin mapping definition in cpu_map.h is described in the wiki here:  
https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h

In your config, the axis C is AXIS_4 and B is AXIS_5. So, to affect the Xmax (pin D2) to Bmin, you must change the values of MIN_LIMIT_PORT_5 and MIN_LIMIT_BIT_5 to:
```
#define MIN_LIMIT_PORT_5 E
#define MIN_LIMIT_BIT_5 4

```

And to affect Ymax (pin D15) to Cmin, change the values of MIN_LIMIT_PORT_4 and MIN_LIMIT_BIT_4 to:
```
#define MIN_LIMIT_PORT_4 J
#define MIN_LIMIT_BIT_4 0

```
@++;
Gauthier

Horan31, 2025-05-18:
Good morning @fra589 
Thanks for all your help, i will what you've mentioned and i will provide you by the results that i will get .

Horan31, 2025-05-22:
Hello @fra589  
I  changed the axes names the A axis with th C axis in config.h and it worked  very fine  , thank you again for solving this issue.

closed 2025-05-23 by fra589